nightscout/AndroidAPS #3301: 3.3.0-dev Build failure:  items queued for execution but none of them can be started

vanelsberg, 2024-04-10:
Android Studio "Menu - > Rebuild project" fails to execute

**AndroidAPS 3.3.0-dev-b**
**Build**: 3bb0705a
**Flavor**: Fullrelease

**Problem description:**
Problem in build definitions _(or is this a local problem?)_

The following actions can be executed successfully:
- Build -> Make Project
- Build -> Generate signed bundle / APK
- Build -> Clean

However, menu action Build -> Rebuild project (Debug) fails to execute, erroring out at the final phase of the build:
*Unable to make progress running work. There are items queued for execution but none of them can be started*

Tried setting op nieuw build environment by cleraing .gradle cache and recloning the project at a different location.
This did not solve the problem.

**Additional info**
*AS About:*
Android Studio Iguana | 2023.2.1 Patch 1
Build #AI-232.10300.40.2321.11567975, built on March 13, 2024
Runtime version: 17.0.9+0--11185874 amd64

**Logs**

(see details attached)

      - Scheduling events:
          - node added to plan: destroyer locations for task group 0, when: scheduled, state: SHOULD_RUN, dependencies: 1, is ready node? false
          - node added to plan: destroyer locations for task group 1, when: scheduled, state: SHOULD_RUN, dependencies: 1, is ready node? false
      - Ordinal groups:
          - group 0 entry nodes: [:buildSrc:clean (SHOULD_RUN)]
          - group 1 entry nodes: [:buildSrc:assemble (EXECUTED)]
          - group 2 entry nodes: [:buildSrc:testClasses (SHOULD_RUN)]
      - Workers waiting for work: 9
      - Stopped workers: 1

FAILURE: Build failed with an exception.

* What went wrong:
Unable to make progress running w

[aserr_build.txt](https://github.com/nightscout/AndroidAPS/files/14931718/aserr_build.txt)
ork. There are items queued for execution but none of them can be started

vanelsberg, 2024-04-10:
Others reporting no problem so likely a local problem. But seeing community ports on similar problems.
@MilosKozak  So will go investigating and report back.

vanelsberg, 2024-04-10:
This is much like: https://issuetracker.google.com/issues/328871352

vanelsberg, 2024-04-11:
Checked building from master: same problem :-|

MilosKozak, 2024-04-29:
It's you local studio installation. I caanot help here

closed 2024-04-29 by MilosKozak

vanelsberg, 2024-04-29:
> It's you local studio installation. I caanot help here

Yes, looks like it.

Still no idea what the problem is, other then that ""rebuild" has problems with some of the unit testing.
Will report back if I find anything of interest.

vanelsberg, 2024-05-03:
@MilosKozak Upgrading to Android Studio Jellyfish | 2023.3.1 (Build #AI-233.14808.21.2331.11709847, built on April 12, 2024)
Solved my problem.

Upgraded by uninstalling while keeping current settings)