nightscout/AndroidAPS #3302: SMB Always and after carbs disabled

rpm5099, 2024-04-13:
I am having recently an error:

`Safety: SMB always and after carbs disabled because active BG source doesn't support advanced filtering`

This is using a G6 sensor with xDrip+ as the collector.  I just noticed this recently and the only thing I could find was #389 and #380, which were not helpful.  Both xDrip and AAPS are re-installed as of 15 minutes ago with the latest prod release versions pulled from both repositories main.  Appreciate any help, I need these two settings enabled or I will have to modify the app which I would like to avoid.  Let me know if there is any additional information I need to provide. 

Thanks to all the devs of this excellent project, I cannot overstate my gratitude.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/14965356/AndroidAPS.log)
[image]

olorinmaia, 2024-04-15:
Is xDrip+ in companion mode? If not, make sure it's setup in line with aaps wiki
https://androidaps.readthedocs.io/en/latest/Hardware/DexcomG6.html#if-using-g6-with-xdrip
https://androidaps.readthedocs.io/en/latest/Configuration/xdrip.html#xdrip-settings

rpm5099, 2024-04-16:
I *think* I figured it out - xDrip was set on a follower to upload treatments via NS instead of through the AAPS client app.  I'm not entirely sure and I wish there was more information in the logs to troubleshoot it but I'm no longer .  Thanks for your response.

closed 2024-04-16 by rpm5099