nightscout/AndroidAPS #3305: Feature / Medtrum Nano : add history

robertrub, 2024-04-19:
Feature request : On the Medtrum app, there is a pump history function. This information is missing in the Medtrum driver in AAPS. 

Something similar to DASH _history button_ could be added to the Medtrum driver. It will be helpful when there are connection problems and the sent insulin amount is not easily verified.

jbr7rr, 2024-04-19:
All insulin related commands are back synced into AAPS history, so I would say that that use case is already covered.

Also Medtrum doesn't have the command structure that Dash had. It has history records.

Let's go over these records and see what we show and don't show:
BOLUS_RECORD, BOLUS_RECORD_ALT > All possible records are synced to AAPS history
BASAL_RECORD, BASAL_RECORD_ALT > Basal modes that AAPS uses are synced to AAPS history, some auto mode records are not synced, but when using medtrum with AAPS this should not be possible to get these records
ALARM_RECORD > Record not synced, but we get notifications on warnings, and it is visible in Medtrum overview TAB
AUTO_RECORD > Not synced, when using medtrum with AAPS this should not be possible to get these records
TIME_SYNC_RECORD > We give a notification when time is synced.
AUTO1_RECORD > Not synced, when using medtrum with AAPS this should not be possible to get these records
AUTO2_RECORD > Not synced, when using medtrum with AAPS this should not be possible to get these records
AUTO3_RECORD > Not synced, when using medtrum with AAPS this should not be possible to get these records

See also:
https://github.com/nightscout/AndroidAPS/blob/5d3e33ce8a0917d0f4130538a4b7c9ec513faeef/pump/medtrum/src/main/java/info/nightscout/pump/medtrum/comm/packets/GetRecordPacket.kt#L88

Notifications are also visible in AAPS history, during implementation we have considered a seperate history screen much like dash.
But given that all relevant information is already synced in some way I don't really see the value of synced these history records.

Another thing is about connection issues, we could add additional notifications, which are then also visible in AAPS history.

To conclude: What medtrum app shows in Pump history is for most parts already visible in the normal AAPS treatments history.

Imo if we want to reconsider a seperate pump history for Medtrum in AAPS I would like to know which information we are currently lacking and what the use case to have it is.

robertrub, 2024-04-19:
Ok, with your explanations, not having a _history_ button seems the logical option.

I'll start using Medtrum on the 29th so I don't have a direct experience with it yet.

I'll close the feature request.

closed 2024-04-19 by robertrub