nightscout/AndroidAPS #3309: Failed bolus by SMS was actually delivered

gh4tf, 2024-04-25:
At 11:24 today, I sent an SMS to bolus for my daughter at school. I immediately received a reply saying the bolus failed, so I sent another text to bolus. This was at 11:25.

Nightscout doesn't update all the time due to connectivity problems at school, so I couldn't check on Nightscout, but then it did update and I saw that the failed bolus had actually been delivered.

Using AAPS 3.2.0.2 Build c31617fd2c-2023.12.13, Combo, G6, xDrip and Xiaomi A2 Lite.

I've uploaded the SMS screenshot, the bolus of 2.5u at 11:24 and 11:25 on Nightscout, a screenshot of the Treatments in AAPS and the two zip files nearest to the time the "failed bolus" occurred; one at 10:27 and 13:13.  

We've been using SMS bolus commands for years and don't think we have ever encountered this before. 

[image]
[image]
[image]
[AndroidAPS._2024-04-25_01-00-20_.4.zip](https://github.com/nightscout/AndroidAPS/files/15119961/AndroidAPS._2024-04-25_01-00-20_.4.zip)
[AndroidAPS._2024-04-25_01-00-20_.5.zip](https://github.com/nightscout/AndroidAPS/files/15119962/AndroidAPS._2024-04-25_01-00-20_.5.zip)

gh4tf, 2024-04-26:
[AndroidAPS._2024-04-25_01-00-20_.docx](https://github.com/nightscout/AndroidAPS/files/15126904/AndroidAPS._2024-04-25_01-00-20_.docx)

MilosKozak, 2024-04-29:
this seems to be combo v2 related @dv1

gh4tf, 2024-04-29:
Thanks @MilosKozak 

Actually, now you mention that, since updating AAPS and using Combo v2 instead of Ruffy, we are getting more frequent bolus delivery failed notifications when we have personally used AAPS to give insulin and know it has definitely been delivered when our daughter is at home. That is what may have happened this time and caused the bolus failed SMS.

closed 2024-05-02 by MilosKozak

gh4tf, 2024-05-02:
Hi, how will I know if there is a reason why this happened please as the issue has been closed?

dv1, 2024-05-02:
> Hi, how will I know if there is a reason why this happened please as the issue has been closed?

I added changes that, among other things, alter how the driver behaves during bolus delivery. The previous behavior may have caused confusion inside the Combo. However, it is not (yet) guaranteed to fix the problem. Please test when you can.

gh4tf, 2024-05-02:
@dv1 thanks. I have just created issue 3318 which shows AAPS screenshots for the bolus delivery failed error. It happened last night. I don't know if you made the changes before then though?

(I created a separate issue, instead of including it in this issue, because this issue was specifically about SMS commands, whereas the new issue was directly on AAPS, although the cause sounds like it is the same.)

dv1, 2024-05-02:
@gh4tf This was merged today morning, so I doubt it. Check your version and git commit hash, and verify that you are really at the very newest dev version.

pontomedon, 2025-01-13:
I installed 3.3.1.0 yesterday, and unfortunately, this issue still persists. Happens on maybe every 10th or 20th bolus delivery:

1. use bolus advisor to enter carbs, calculate insulin, deliver
2. the pump connects, bolus delivery progress bar runs all the way to 100%, but then something goes wrong and
3. a "Bolus delivery failed" announcement is logged
4. both insulin and carbs are missing from the treatments, only the calculation is there
5. Bolus is imported from pump on next history sync, carbs are still missing.

Please let me know if I can provide any logs or how I could debug this (professional Android developer here, so no problem to get technical).