nightscout/AndroidAPS #3318: Failed bolus error message when bolus was delivered

gh4tf, 2024-05-02:
At 16:54 yesterday, I gave my daughter 2.1u of insulin directly using AAPS and got the "Bolus reported an error" message (please see screenshot), but I know that the bolus was definitely delivered by the Combo pump because it delivers insulin in 0.1u increments and I count the little noises the pump makes with each one.

I have included a screenshot of the treatments showing 2.1u was delivered at 16:54 and the main screen of AAPS showing bolus delivery failed in orange on the graph.

I think this is what happened when I received a "bolus failed" text when sending an SMS command recently. I created issue Failed bolus by SMS was actually delivered #3309 for that.

Using AAPS 3.2.0.2 Build c31617f 2023.12.13, Combo, G6, xDrip and Xiaomi A2 Lite. No longer using Ruffy.

@dv1 

[Log Export.txt](https://github.com/nightscout/AndroidAPS/files/15186441/Log.Export.txt)

[image]
[image]
[image]

dv1, 2024-05-02:
As I said [here](https://github.com/nightscout/AndroidAPS/issues/3309#issuecomment-2090023154), try newest dev.

closed 2024-11-13 by MilosKozak

pontomedon, 2025-01-14:
@MilosKozak as mentioned [here](https://github.com/nightscout/AndroidAPS/issues/3309#issuecomment-2587085926), this issue is unfortunately not fixed. It started occurring with the new Combo driver (which otherwise works great!), and is still present on `3.3.1.0`.

Pretty sure #3318 and #3309 are duplicates, but one of them should be re-opened, I think.

Der-Schubi, 2025-01-14:
Can confirm, I have this problem every few days.

dv1, 2025-01-15:
@Der-Schubi If you can, do the following:

Open the file `pump/combov2/comboctl/src/commonMain/kotlin/info/nightscout/comboctl/base/PumpIO.kt`.

In lines 1662, 1694, 1728, 1733, 1742, 1774, 1807, 1812 change `VERBOSE` to `DEBUG`, like this:

from:

    logger(LogLevel.VERBOSE) { "Starting background CMD ping heartbeat" }

to:

    logger(LogLevel.DEBUG) { "Starting background CMD ping heartbeat" }

Then, when the issue happens again, immediately post logs here.

Der-Schubi, 2025-02-04:
> Any news here?

Not yet . Got no errors since. May be a timing issue, resolved by the debug logging?

pontomedon, 2025-02-05:
Had it like 3 or 4 times, but was never fast enough to connect ADB. My wife's the patient, so I have limited opportunities where she injects while being near my computer. I think I'll set up wireless ADB and just connect whenever she's in the apartment :P

Der-Schubi, 2025-02-05:
> Had it like 3 or 4 times, but was never fast enough to connect ADB.

You can export the logs without ADB, directly in AAPS. In the menu under maintenance is an option to send logs by email.

pontomedon, 2025-02-05:
> You can export the logs without ADB, directly in AAPS

damn, didn't know that. will do when it happens the next time!

dv1, 2025-02-16:
I already posted this in Discord, but for higher visibility, I also mention it here:

I added a `comboctl-dev` branch to my https://github.com/dv1/AndroidAPS fork. If you can, try that out and let me know if this fixes the issue for you. Due to the nature of the issue, reproducing it can be very difficult, hence my request.

Der-Schubi, 2025-02-16:
I think I captured one of the errors today, but had no time to look into the logs for now. I'll have a look and post the logs tomorrow.

Der-Schubi, 2025-02-17:
@dv1 This was just insulin, no carbs this time. But would I have entered carbs, they would have been deleted.
Bolus was 6U, delivered were the full 6U, but there was a disconnect after ~3U.
```
16:58:39.035 [DefaultDispatcher-worker-10] D/PUMP: [Pump.setState():3620]: [Pump] Setting Combo driver state:  old: info.nightscout.comboctl.main.Pump$State$ReadyForCommands@4ce59f3  new: ExecutingCommand(description=info.nightscout.comboctl.main.Pump$DeliveringBolusCommandDesc@3b63355)
16:58:39.037 [DefaultDispatcher-worker-10] D/PUMPCOMM: [PumpIO.stopCMDPingHeartbeat():2200]: [PumpIO] Background CMD ping heartbeat stopped
16:58:39.037 [DefaultDispatcher-worker-10] D/PUMPCOMM: [PumpIO.startCMDPingHeartbeat():2194]: [PumpIO] Starting background CMD ping heartbeat
16:58:39.065 [DefaultDispatcher-worker-1] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.plugins.sync.nsclientV3.workers.DataSyncWorker Data {}
16:58:39.243 [DefaultDispatcher-worker-6] D/PUMP: [ApplicationLayer.parseCMDReadErrorWarningStatusResponsePacket():1894]: [ApplicationLayer] Parsing CMD_READ_ERROR_WARNING_STATUS_RESPONSE packet
16:58:39.244 [DefaultDispatcher-worker-6] D/PUMP: [ApplicationLayer.parseCMDReadErrorWarningStatusResponsePacket():1897]: [ApplicationLayer] Error/warning status: CMDErrorWarningStatus(errorOccurred=false, warningOccurred=false)
16:58:39.244 [DefaultDispatcher-worker-6] D/PUMP: [Pump$deliverBolus$3.invokeSuspend():3590]: [Pump] Checking if there is enough insulin in reservoir; reservoir fill level: 141 IU; bolus amount: 6.0 IU(rounded: 6 IU)
16:58:39.244 [DefaultDispatcher-worker-6] D/PUMP: [Pump$deliverBolus$3.invokeSuspend():3593]: [Pump] Beginning bolus delivery of 6.0 IU
16:58:39.244 [DefaultDispatcher-worker-6] D/PUMPCOMM: [PumpIO.stopCMDPingHeartbeat():2200]: [PumpIO] Background CMD ping heartbeat stopped
16:58:39.245 [DefaultDispatcher-worker-6] D/PUMPCOMM: [PumpIO.startCMDPingHeartbeat():2194]: [PumpIO] Starting background CMD ping heartbeat
16:58:39.774 [DefaultDispatcher-worker-6] D/PUMP: [ApplicationLayer.parseCMDDeliverBolusResponsePacket():1936]: [ApplicationLayer] Parsing CMD_DELIVER_BOLUS_RESPONSE packet
16:58:39.775 [DefaultDispatcher-worker-6] D/PUMP: [ApplicationLayer.parseCMDDeliverBolusResponsePacket():1939]: [ApplicationLayer] Bolus started: true
16:58:39.776 [DefaultDispatcher-worker-6] D/PUMP: [Pump$deliverBolus$3.invokeSuspend():3599]: [Pump] Waiting until bolus delivery is complete
16:58:39.778 [DefaultDispatcher-worker-6] D/PUMPCOMM: [PumpIO.stopCMDPingHeartbeat():2200]: [PumpIO] Background CMD ping heartbeat stopped
16:58:40.241 [DefaultDispatcher-worker-1] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1930]: [ApplicationLayer] Parsing CMD_GET_BOLUS_STATUS_RESPONSE packet
16:58:40.243 [DefaultDispatcher-worker-1] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1933]: [ApplicationLayer] Bolus status: CMDBolusDeliveryStatus(bolusType=STANDARD, deliveryState=DELIVERING, remainingAmount=59)
16:58:40.703 [DefaultDispatcher-worker-2] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1930]: [ApplicationLayer] Parsing CMD_GET_BOLUS_STATUS_RESPONSE packet
16:58:40.705 [DefaultDispatcher-worker-2] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1933]: [ApplicationLayer] Bolus status: CMDBolusDeliveryStatus(bolusType=STANDARD, deliveryState=DELIVERING, remainingAmount=58)
[...]
16:58:52.849 [DefaultDispatcher-worker-10] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1930]: [ApplicationLayer] Parsing CMD_GET_BOLUS_STATUS_RESPONSE packet
16:58:52.851 [DefaultDispatcher-worker-10] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1933]: [ApplicationLayer] Bolus status: CMDBolusDeliveryStatus(bolusType=STANDARD, deliveryState=DELIVERING, remainingAmount=34)
16:58:53.320 [DefaultDispatcher-worker-7] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1930]: [ApplicationLayer] Parsing CMD_GET_BOLUS_STATUS_RESPONSE packet
16:58:53.324 [DefaultDispatcher-worker-7] D/PUMP: [ApplicationLayer.parseCMDGetBolusStatusResponsePacket():1933]: [ApplicationLayer] Bolus status: CMDBolusDeliveryStatus(bolusType=STANDARD, deliveryState=DELIVERING, remainingAmount=33)
16:59:02.755 [DefaultDispatcher-worker-10] D/PUMP: [TransportLayer$IO$startInternal$1.invokeSuspend():1263]: [TransportLayer] Caught Combo exception in receive loop: info.nightscout.comboctl.base.ComboIOException: Could not read data from device with address 00:0e:2f:XX:XX:XX
16:59:02.761 [DefaultDispatcher-worker-10] D/PUMP: [TransportLayer$IO$startInternal$1.invokeSuspend():1266]: [TransportLayer] Combo exception stacktrace: info.nightscout.comboctl.base.ComboIOException: Could not read data from device with address 00:0e:2f:XX:XX:XX
	at info.nightscout.comboctl.android.AndroidBluetoothDevice.blockingReceive(AndroidBluetoothDevice.kt:216)
	at info.nightscout.comboctl.base.BlockingComboIO$receive$2.invokeSuspend(ComboIO.kt:90)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:100)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:113)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:89)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:586)
[... cut ...]
17:00:14.729 [DefaultDispatcher-worker-3] D/PUMP: [Pump.scanHistoryDeltaForBolusToEmit():3676]: [Pump] Found a last bolus in history delta; details: LastBolus(bolusId=1093169, bolusAmount=60, timestamp=2025-02-16T15:58:17Z); now: 2025-02-16T16:00:14.729059Z; lastBolusInfusionTimestamp: 2025-02-16T15:58:17Z -> bolusTimestamp: 2025-02-16T15:58:17Z
[...]
17:00:15.112 [main] D/CORE: [MaintenancePlugin.sendLogs():76]: zipFile: AndroidAPS_LOG_1739721615016.log.zip
17:00:15.119 [main] D/CORE: [MaintenancePlugin.zipLogs():146]: creating zip AndroidAPS_LOG_1739721615016.log.zip
```

Der-Schubi, 2025-02-22:
@dv1 See new logs in Discord combo-v2 channel.

docx1, 2025-09-20:
I can confirm that this error is happening again. I thought that it was battery related, changed battery still the same error occurred.