nightscout/AndroidAPS #3321: Pump doesn't automatically reconnect like it used to

GoudekettingRM, 2024-05-06:
Eversince I updated to v3.2.0.4 every time I change my cartridge or battery, AAPS does not automatically reestablish connection. It used to do that, but now I have to go to my pump tab and click "refresh" in order for it to link up again and continue working. If I don't, it just doesn't engage and it has caused my BGs to go out of wack because the closed loop is not engaged. 

It happens every time when I change the battery of my pump or put a new cartridge in. 

Pump: Accu-chek Insight
AAPS version: 3.2.0.4

I don't know if this has been an intentional change or that it's a bug, but I'd love to have it reconnect automatically.

parapenT1sta, 2024-05-22:
@dv1 ?

MilosKozak, 2024-11-13:
@Philoul

Philoul, 2024-11-14:
Very strange, AFAIK there were no change within Insight driver in current master compare to previous versions.

The only recent update of Pump driver has been made in latest dev to migrate code to kotlin, and during development I checked the automatic reconnection after cartridge change, and it still work as expected in dev (will check again next time...).

The only reason that could explain this behaviour is if you forget to restart the pump after cartridge change, then AAPS cannot reconnect to the pump, but if you confirm pump was restarted manually, then I have no ideas... 🤔

GoudekettingRM, 2024-11-14:
Interesting. What do you mean with restart? The pump always reinits itself with a full test, but restarting by turning it on and off was never required before..

Philoul, 2024-11-14:
When cartridge is empty or when battery is off, pump stop to inject insulin.
So you have to start the pump from pump menu after cartridge (or battery) change to "re-start" insulin delivery.
Pump driver can connect to the pump only when pump is "started", so it cannot start a pump remotely, this must be done from pump.

On cartridge change, pump UI ask you first if you want to fill tube (generally we should accept), then if you want to fill canula (here you can accept if you synchronize site change and cartridge change, or refuse if site and cartridge is not synchro), and finally ask you if you want to restart pump... Here you should accept (mandatory to start insulin delivery and allow pump connection to the pump...).

GoudekettingRM, 2024-11-14:
ohh, that's what you mean. Yeah, of course, that's part of the cartridge change in my eyes haha.

Philoul, 2024-11-14:
> but restarting by turning it on and off was never required before..

Yes it was, this can't be done remotly.

But the pump synchro within Insight plugin (or automatically on next pump connection request) can only connect to the pump if pump is started.

GoudekettingRM, 2024-11-14:
Yeah, but the pump is started (i.e. I can bolus through the pump itself), but AAPS does not connect. I have to manually go to the status tab and click refresh for it to connect again.

Philoul, 2024-11-25:
I can't reproduce your issue (but i'm in latest dev so not exactly the same config than yours...)
During the last 2 weeks, I managed the reservoir change each time it's 100% empty (so most of the time it's when pump is trying to inject a Bolus, a SMB, or during a TBR).
- I confirm the alert on phone
- I changed reservoir, managed prime / fill of tube and cannula
- Restart my Pump

During all these steps I could see in the bottom of AAPS Overview, the connection message trial... 
Then after pump restart and screen is black again on pump (pump cannot communicate with BT when pump screen is on), then pump connection is again established and current pending action is finished

Once I wait too long between the alert and the change (you only have 15 min if you want current pending command is resumed), then pump reconnected automatically but without the "Resume" so previous bolus was partially injected (but an alert on pump UI)

So according to me everything worked as expected (major update between current master and dev is migration of code from java to kotlin), but this migration is neutral concerning communication between Pump and phone...)
Could you provide log files during reservoir change and at least 1 hour or more after the end.

Other possibility, do you wait enough time after reservoir change (if alert and reservoir is changed without any pending bolus or SMB command, it's possible AAPS wait until a new command should be sent to the pump, so it can be several "loop run" later...)

Philoul, 2024-11-25:
I talked about this message in the bottom "Connecting for xxx s".
Here loop was running. but during reservoir change, you should have this message all the time in parallel of red loop icon, until AAPS is able to communicate again with the pump...

[image]

Philoul, 2025-06-07:
Do you still have this issue with latest release?
I'm using Insight pump with hundred of cartridge changes, battery changes and never was able to reproduce it.
Communication needs at least to have a black screen on the pump, so immediatly after battery/cartridge change or after any user action on pump, you will have to wait until pump screen turn black (about 1 min according to user timeout).
Then nothing else to do, on the next loop run with a required update on pump, communication is automatic without any additional manual action from user...

closed 2025-11-23 by MilosKozak