nightscout/AndroidAPS #3324: AAPS 3.2.0.3  Samsung Galaxy Watch 5  not connected

AngelAlex1975, 2024-05-06:
There is AAPS version 3.2.0.3 and there is a Samsung Galaxy Watch 5. I installed the Wear-aapsclient-3.2.0.3 application on the watch.
At the same time, on the phone in AAPS on the SMART WEAR tab - (the watch is not connected). Is it possible to somehow connect the watch and phone?

MilosKozak, 2024-05-06:
GW5 works fine with AAPS
make sure you see watch properly connected in Wear app on phone and you signed both apks with the same keys

AngelAlex1975, 2024-05-06:
The watches in "Galaxy Wearable" are connected without error. aapsclient +wear-aaps is signed with one key. Are there more detailed instructions for pairing AAPS + Wear? This question is also raised in the local telegram group.

AngelAlex1975, 2024-05-06:
I took another Samsung A50 phone and reset the clock to default. Installed wear-aapsclient 3.2.0.3/ Connected Galaxy Wearable and Still AAPS on Samsung Galaxy 5 does not work.

AngelAlex1975, 2024-05-06:
3.2.0.4 also tried

olorinmaia, 2024-05-06:
AAPSClient and AAPS are two different apps. So if you got AAPS on your phone you need AAPS wear apk that need to be built in android studios with same keystore as AAPS apk.

AngelAlex1975, 2024-05-07:
I deleted the app on both my phone and watch. Installed 
app-aapsclient-release_3.2.0.3
wear-aapsclient-release_3.2.0.3
 from your repository.
 Everything worked.
 Thanks a lot!