nightscout/AndroidAPS #3326: Unable to connect to Nightscout -No error in log

FamouZzE, 2024-05-06:
I am unable to connect my APS to my self-hosted Nighscout Server. It worked on a previous phone with a previous version of APS. I can access Nighscout site from phone directly via Browser and my xdrip+ Installation is also able to upload to Nighscout. 
The Log only shows the following two lines:
14:05:13.122 [RxCachedThreadScheduler-37] D/NSCLIENT: [NSClientPlugin$onStart$5.accept():112]: ● NSCLIENT destroy
14:05:13.124 [RxCachedThreadScheduler-37] D/NSCLIENT: [NSClientPlugin$onStart$5.accept():112]: ● NSCLIENT do connect 

In the UI APS shows also only the two lines and state never changes from connecting.
[image]

I am running APS Version 3.2.0.4 (Build: 5d3e33ce8a-2024.05.06) what seems odd to me is the Nighscout version "unknown" but maybe this is only reported if I could connect correctly.
[image]

What I Already tried without success:
- Clearing Data and Cache APS
- Clean new Installation of APS
- Change Nighscout default role to admin
- Clear storage of Nighscout DB
- Restart Nighscout
- Change Port from 5423 to standard 443

MilosKozak, 2024-05-06:
what is your NS version?

MilosKozak, 2024-05-06:
then read release notes

closed 2024-05-06 by MilosKozak

FamouZzE, 2024-05-06:
Sorry my bad, after update to version 15.0.2 I do still get a 401 error in the log, but URL is correct and I checked token also multiple times. Am I missing anything?