nightscout/AndroidAPS #3334: Crash across multiple versions related to medtronic driver (seems resettable)

PedanticAvenger, 2024-05-10:
Pump: Medtronic 722
AAPS Versions: 3.2.0.2/3.2.0.4/3.3-dev-B

Description
At 0700 today my app began crashing repeatedly.  It would occur at the end of the data download from the pump and would happen every time the initial connect/download would complete.  Upgrades of app from 3.2.0.2 -> 3.2.0.4 -> dev did not change behavior.
Moving to virtual pump for a while then moving back to Medtronic driver interrupted the behaviors and seemingly normal function resumed.

Various logs from the impacted periods included.

Reporting bugs
[AndroidAPS_LOG_1715360030719.log.zip](https://github.com/nightscout/AndroidAPS/files/15278752/AndroidAPS_LOG_1715360030719.log.zip)
[AndroidAPS_LOG_1715357887860.log.zip](https://github.com/nightscout/AndroidAPS/files/15278753/AndroidAPS_LOG_1715357887860.log.zip)
[AndroidAPS_LOG_1715347051272.log.zip](https://github.com/nightscout/AndroidAPS/files/15278754/AndroidAPS_LOG_1715347051272.log.zip)

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2024-11-13:
@andyrozman

MilosKozak, 2026-10-11:
Your log shows the crash came from a pump suspend that started in the same second as a temp basal, and AAPS tried to end that temp basal with zero length. This case is now handled (fixed in 3.4.2.3 and 4.0.0). Please reopen with a new log if it happens again.

closed 2026-10-11 by MilosKozak