nightscout/AndroidAPS #3335: AAPS Client Limitations, thoughts

rpm5099, 2024-05-12:
I am sure that I am bringing up a topic that has been discussed at great length -  with many differing opinions and points of view.  I apologize that I was not a party to previous discussions, and if these are limitations of developer time and effort or legal matters, then I apologize up front.  My goal is to determine if I am missing *anything* in terms of the trade-offs between excellent DM management and safety when it comes to the AAPS client.  BLUF - why are certain features are included in the client app and others excluded?  I do not believe there are any technical hurdles that would prevent the AAPS client from both (A) displaying the same information as the master or (B) having the same functions and control as the master.  

The use cases for AAPS client that I can think of are:
- Parent(s) managing a child (or anyone managing an insulin pump for another person for any reason)
- Backup phone in case of loss/damage?
- Remote monitoring by a 3rd party (in a read only situation?)

None of these situations would preclude the AAPS client mirroring exactly the function and display of the master in terms of safety - in fact very much the opposite.  While the SMS features do allow remote commands, including bolus, I am not willing to use it in most scenarios because I do not have access to the information about why previous decisions have been made or any details of the calculations involved.

It appears that at least the following functions are not available in AAPS client:
  - Bolus
  - Preferences
  - Loop Status/settings
  - Pump settings, history and configuration
  - SMB calculations and status
  - Temporary Basal modification
  - Maintenance (Settings export - clearly needs to be separate from the client app itself)
  
The *one area* that I believe there is reason for concern over is possible confusion between AAPS client and master on the same phone (controlling two different people), however there a lot of options for mitigating that risk - additional authentication (2FA as used in SMS), notifications and/or confirmation.  Authentication/verification to enable full control should be implemented over feature removal in the client.

I am diabetic and I have a 5 year old daughter with T1DM, and I cannot possibly state how grateful I am for the existence of this project and the example it sets.  I have spent a great deal of time and money on phones, remoting software, rooting, bypassing security features, etc. so that I have full control and visibility - and yet it all fails with poor signal as not enough bandwidth to successfully remote.  Stating the obvious, my daughter is not in a position to be aware of or make any decisions regarding her insulin pump, and I am struggling to think of a scenario where someone is capable of making insulin pump decisions and at the same time has someone else viewing/controlling their insulin pump via the AAPS client.  I appreciate any insight.

I truly want to thank all the devs of this this excellent capability.    I'm not strong at java, capable with python/C/x86 assembly, but certainly willing to contribute.

olorinmaia, 2024-05-13:
I agree to some extent that some of these functions would be handy to have in AAPSClient, but there may be technical limitations for how this is handled through NSClient API that limit what can be done, can be better answered by @MilosKozak . There is also a concern about security and potential overdosing when more than one device can administer boluses directly.

**"SMB calculations and status"** is available in Nightscout and AAPSClient by clicking on OAPS / OpenAPS button in overview screen. It's a summary though, and doesn't contain all the info available in debug-screen in SMB-tab in AAPS, but for my part, it got all I need, which is mainly calculated "insulinReq" and SMB size.

**"Temporary Basal modification"** is handled through oref, and can be affected by setting temporary target and shouldn't be needed for interference from AAPSClient.

**"Preferences"
"Loop Status/settings"
"Pump settings, history and configuration"**
None of these are needed to be configured or viewed so often that syncronization between AAPS and AAPSClient is needed imo. An option in those cases where child is far away and you want to change things is Teamviewer.

Did you know that Wear OS got support for communication with phone through internet/esim/wifi? :) This is a function that is automatically enabled on Samsung watches and when bluetooth connection is not available the watch can sync over Google Cloud with phone. I actually disable bluetooth on the watch connected to AAPS phone and only communicate through Google Cloud. That way I can use AAPSClient on my phone and AAPS wear to handle my son's AAPS. That way I can give bolus if needed, allthough he isn't in bluetooth-range. It works very well for us. With this setup, it isn't really needed with SMS commands.

More info about Google Cloud Sync Service and remote control via wear OS:
https://developer.android.com/training/wearables/data/data-layer#connectivity
https://androidaps.readthedocs.io/en/latest/remote-control.html#remote-control-of-aaps

MilosKozak, 2024-05-14:
Technical limitations come from NS which is man in the middle of sync.
Communication through Google cloud is possible but it means to use tha same Google acccount on all devices which is not a common case

olorinmaia, 2026-09-23:
There is new master/client sync coming in AAPS 4.0.0. Allows you to change most settings you'd want to alter like those for SMB algorithm  Can also deliver bolus directly. Quick wizard sync. Automation sync. Scenes sync. Basal/TBR can be changed. Certain configuration can be changed, like Smoothing type. Running mode can be changed. 

If there isn't anything else you'd think would be useful in this new sync you can consider closing this issue :)

rpm5099, 2026-09-24:
> There is new master/client sync coming in AAPS 4.0.0. Allows you to change most settings you'd want to alter like those for SMB algorithm Can also deliver bolus directly. Quick wizard sync. [...]

That is fantastic news.  I will let you know as soon as I have a chance to check it out.

rpm5099, 2026-10-26:
> There is new master/client sync coming in AAPS 4.0.0. Allows you to change most settings you'd want to alter like those for SMB algorithm  Can also deliver bolus directly. Quick wizard sync. [...]

@olorinmaia Hi there - I had a chance to review/test the latest changes and it is a major improvement - I really appreciate it.  I did find a few issues that I am assuming you would rather have created separately.  Let me know and I can do that.  

The most pressing one is the usage of the Android hardware KeyStore API (StrongBox on some devices) - the decrypted data needs to be cached because there is a hardware limit on the rate that the API can be called and it competes with everything else running on the device.  They want it to be called with a backoff using `RETRY_WITH_EXPONENTIAL_BACKOFF`.  This results in failed commands from the client with no retry/acknowledgment and does not implicate the keystore in the logs.  This includes the "stop bolus" feature.  The limit is documented in the [KeyMint interface](https://android.googlesource.com/platform/hardware/interfaces/+/refs/heads/main/security/keymint/aidl/android/hardware/security/keymint/IKeyMintDevice.aidl), the retry policy in [KeyStoreException](https://developer.android.com/reference/android/security/KeyStoreException), and this is the [actual call to the API](https://github.com/nightscout/AndroidAPS/blob/3905d352ddaa18bd29edb5604b5c3b501a86dcd9/implementation/src/androidMain/kotlin/app/aaps/implementation/protection/SecureEncryptImpl.kt#L325) in the AAPS code.

There are also a few other features not enabled via the client api's that would be really helpful:

- Deleting a treatment in Nightscout does not delete it from AAPS - ideally these would be bidirectionally synchronized
- Reporting GPS location to nightscout/client and locating audible alerts would be really helpful with children - there are other solutions but they have their own issues
- Silencing of pump alarms/beeping from client - I use Omnipod Dash so that's what I tested with
- the permission nags should probably be reduced to bluetooth, especially if all notifications are sent to nightscout/client
- edits to the currently running profile should have the option to take effect immediately without a profile switch, that very often goes unnoticed
- exposing to nightscout/client more raw pump information would be very helpful, including the last successful Bluetooth connection

I am happy to report my other findings in new issue(s), let me know what works best for you.  Do you want this issue closed now or when the changes are released to the production version?  Thanks again.