nightscout/AndroidAPS #3336: After replacing the battery in the Accu-Chek Combo, AAPS cannot automatically reconnect.

birdfly, 2024-05-14:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

When the pump battery is depleted, it will automatically stop infusion. AAPS will display that the pump has stopped running. After replacing the battery, you need to manually refresh the pump to reconnect. If this operation is not performed, AAPS will remain in a suspended state.

parapenT1sta, 2024-05-22:
@dv1

jotomo, 2024-08-24:
AAPS automatically connects to the pump if the last connection was more than 15 minutes ago (see _KeepAliveReceiver.checkPump()_).
Since AAPS can't know when a battery or cartridge change is completed, the user can either inform AAPS via a manual refresh, or simply wait for AAPS do so as described above (also if the user forgets to perform a manual refresh).

So, while not obvious to the user, AAPS does handle this case.

closed 2024-11-13 by MilosKozak