nightscout/AndroidAPS #3340: DASH/EROS : Feature / add export settings button

robertrub, 2024-05-15:
It would help people to be "more serious" with backups if there was an _export settings_ button shown at the end of the pod activation sequence.

kenzo44, 2024-05-18:
We can also update the text..."x days since last aaps back up. Please back up aaps to not lose settings"

robertrub, 2024-05-18:
You need to export after each pod activation. If you have a phone problem, do a new installation and then import old settings, not only you'd lose your active pod but also stress trying to deactivate a pod that AAPS can't reach...

It should not be let to believe that doing backups "whenever you have time" is an option ;)

MilosKozak, 2024-11-13:
Automated exports added in 3.3

closed 2024-11-13 by MilosKozak