nightscout/AndroidAPS #3342: Zlyhanie ochrany podať opätovne inzulín na dialku počas 20 min.

Simona0710, 2024-05-17:
Dobrý deň zasielam podklady k zlyhaniu ktoré sa nám stalo 15.05.2024 na pumpe Dana IR s AAPS v najnovšej verzii 3.2.0.4.
Posielala som na diaľku inzulín neskôr ako boli sacharidy nakoĺko olovrant bol 14:15 z hypa. Prvý bolus podávaný 14:52 - so spätnou sms o chybe a poslený bolus bolo SMB o 13:46. Viditeľné zo screenu z SMS. Snažila som sa podať tú prvotnú dávku ktorú mi clientská app neukazovala v grafe a nebola ani v grafe dieťaťa. Bol problém v komunukácii nakoľko bola rozpojená a opatovne sa nevedela spojiť no nevykázala pumpa nedostupná. Učiteľka opakovanie nerešpektuje pokyny a vyberá mobil  z kapsičky dieťaťa.

Po treťom pokuse chýb som si cez BG vyžiadala spojenie info z pumpy kde IOB bolo 0,51U  a prišla dalšia chyba.  Išla som priamo po neho hneď som vypýtala mobil a dieťa v mobile stále bolo len 35g sacharidov vidieť. Cez 3 bodky som ukončila app a zapla nanovo a v tom do grafu aj do rodičovskej app zapísalo všetky bolusy ktoré vrátilo ako chybné a boli riadne vydávkované. Dokonca posledný 4 pokus som zadala pred vypnutím app cez kalkulačku lebo som tam nevidela k 35g olovrantu žiadny inzulín.
Keď to všetko ukázalo a aktívny inzulín v pumpe bol 11U okamžite sme išli domov podali sme kopu cereálii s mliekom a sladkú vodu nezabralo klesanie delta po -2,5 - podali sme celý glukagón do svalu a zvádli sme to veľmi dobre.  Preskúmajte podklady a kedykoľvek ma môžete kontaktovať. 
[AndroidAPS_LOG_1715695405232.log (1).zip](https://github.com/nightscout/AndroidAPS/files/15351561/AndroidAPS_LOG_1715695405232.log.1.zip)
[image]
[image]
[image]
[image]
[image]

Simona0710, 2024-05-17:
[image]

Philoul, 2024-05-20:
@Simona0710 can you please edit issue title and issue description to translate everything in english?

vanelsberg, 2024-05-20:
@Simona0710 Hint: you can try https://translate.google.com
Seems like a dangerous situation your child experienced. You handled well! Not familiar with your setup, can't tell what could have caused your problem🤔 

To help dev's evaluate the problem better, please add some specification on your setup and how you monitor your child remotely? You may also want to connect to the Discord #androidaps-parents channel to discuss what happend?

MilosKozak, 2024-11-30:
Unfortunately this has no solution. The primary reason is bad communication with pump maybe because of distance between pump and phone.
In this case phone was able to to initiate a connection to the pump and start bolus. Then connection was broken and AAPS has to report unsuccessful bolus.
I improved error message anyway https://github.com/nightscout/AndroidAPS/commit/03377bc1dc8b6b3c0e105e646ff1ca6ff7c2cc5b

closed 2024-11-30 by MilosKozak

Simona0710, 2024-12-02:
Ďakujeme ❤️
Simona 
DOSIL J s.r.o.
Terany 177, 96268
IČO: 53707354
IČ DPH: SK2121476302
www.dosiljsro.sk
tel.:0903149061
***@***.*** alebo
***@***.***

> Od: "Milos Kozak" ***@***.***>
> Komu: "nightscout/AndroidAPS" ***@***.***>
> Dátum: 30.11.2024 22:24

Unfortunately this has no solution. The primary reason is bad communication with pump maybe because of distance between pump and phone.
In this case phone was able to to initiate a connection to the pump and start bolus. Then connection was broken and AAPS has to report unsuccessful bolus.
I improved error message anyway 03377bc

—
[
{
"target": "https://github.com/nightscout/AndroidAPS/issues/3342#issuecomment-2509390109",
"url": "https://github.com/nightscout/AndroidAPS/issues/3342#issuecomment-2509390109",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]