nightscout/AndroidAPS #3344: Medtrum : End of pod AAPS alert error/feature request ?

robertrub, 2024-05-20:
Medtrum Nano. I can't hear the pod alerts. I need AAPS alerts too.

The pod end alert is set to 72 hours. I don't know if the pod beeped. 

[image]

No (AAPS) alert 3 days after the pod's end (maybe the pod beeped...). The AAPS alert arrived at 3 days + 8 hours...

[image]

It would be good to have the 72 hours and 72+8 hours alerts in AAPS too

The logs (to see if the pod beeped)

[image]

jbr7rr, 2024-05-25:
Have you considered automations to setup additional alerts?

robertrub, 2024-05-25:
Very good idea, thanks. Didn't know that option existed. I'll try.

closed 2024-06-07 by robertrub