nightscout/AndroidAPS #3349: Why does NS have to be secure HTTPS for writing only ?

sircsaba, 2024-05-25:
I use NS on my RPI4 with HTTPS disabled for ease of use and flexibility. Why do you FORCE us to use HTTPS for writing only ? Why not just have the option do choose for ourselves. I'm old enough to understand the risks etc ... At least, can you document where I can change the calling routine in the library that you use and I will do the mods myself ?

I'm sure that I'm not the only person that has issues using SSL certificates etc ... Especially when it comes to a home server situation. I understand the security between AAPS and the pump for control purposes. 

The reason why we like Open Source is the ability to modify as we feel fit and not be bound my corporate systems built for their needs more than the needs of their clients ! This will help advance systems in the future !

You can email me directly if posting a response isn't appropriate : dfw@inforap.com

Don't get me wrong, I appreciate all the effort you put into this project.

qq254344379, 2024-05-27:
I think http with certificate is necessary. ns provides not only uploading, but also downloading. High-level security settings are more beneficial to use.

vanelsberg, 2024-06-03:
> Not if you need only to write to NS only. I use NS as a viewer for the data.
I think you underestimate the risks in doing such.

Using secure HTTPS for modern webclients is standard for good reasons even when deploying on your home network.

When hosting a web application like NS on a homeserver in your home network you should prefer HTTPS. Implementing access over HTTPS should not complicated and most homeservers have this capability build-in. If not, invest some effort into deployment using a self signed or free Let's encrypt certificate (Check doc's or Google?).

MilosKozak, 2024-06-03:
leaving HTTP is not only a case of entrerprise. HTTPS is just common safety. Along with NGINX and ACME it's easy to achieve even in home use

closed 2024-06-03 by MilosKozak

sircsaba, 2024-06-03:
Don't worry, I know all of this. My point is that the user should decide and not someone else. Its like we all know that smoking is bad for your health but you are an adult and are willing to take the risk, then go ahead ... Like driving on the autobahn at 160km/h versus driving 100km/h. You are informed and you make your decision accordingly.

Same thing with software, there are in many instances warning that this software can do harm to your computer, do you accept the risk. Yes I do, so I continue, like we all do in Windows and other OS ...

If we start making decisions for others then we are no better than big Pharma who are also trying to control the narrative. This is my point.

This has nothing to do with your software or the quality of it of course.

BTW, if you think that hiding behind a secured site you are secure ... Think about ! 

All of this because all I want to do with AAPS and NS is for AAPS to write to my NS and nothing else. But some people have decided for me that I shouldn't be doing it. After over 40 years in the software development career, sometimes things never change. 

Good day and Cheers to everyone. Keep safe and wear your seat belts, just a suggestion ...

MilosKozak, 2024-06-03:
making it configurable (disable https if it doesn't work for me) would lead to situation where non skilled users would just disable it as a resolution without evaluating the risk(s)
IT professional can change the code and will understand the risk or will use VPN instead for example 
optimal solution from my side

sircsaba, 2024-06-03:
No worries, I have already started changing the code but I have to concentrate more on the health issues than this for now. Also, things are moving quickly in the industry and depending on where you live and insurance policies using AAPS as a primary source might prove to be problematic. It would have been nice to evaluate it but, c'est la vie ! 
Keep up the good work and open minds.

sircsaba, 2024-06-03:
Can you quickly tell me which libraries you use for https communications please. This will make it quicker for me in the future when I have time... You can send it to my email : jeff at inforap dot com

vanelsberg, 2024-06-04:
> which libraries...

You will need a frontend like **NGINX** or **Treafic** to handle HTTPS with a certificate. For a certificate you can use "Let's Encrypt" which is totally free (https://**letsencrypt.org)**.

How, depends on your deployment.

My prefred method, easiest would be using Docker, running on a VM or Raspberry Pi 4 (see f.i https://gist.github.com/Andries-Smit/daac75cd4c06af78cde68c5dec941705). All you need is to install Docker and use the projects dockercompose file.

Fore deployment to a NAS like Synology or QNAP see https://nightscout.github.io/nightscout/advanced/#building-nightscout-diy-in-a-virtual-server.

In case you like to deploy to a local Ubuntu/Linux server, just search the internet. Lots of examples, f.i https://gist.github.com/johnmales/1b3c927f2a56aae640b4b2cd0298b1e7