nightscout/AndroidAPS #3351: Dash resovouir empty, No Automated actions

DrunoL, 2024-05-27:
When the resovoir of the Pod is empty, but the pod ist not expired, no Action ist taken automatcally, so the bloodsugar rises without information. 

To say: The best App ever coded, thank you!

[image]

[image]

Philoul, 2024-06-01:
For me it looks more as a pump issue that should be raised in Pump manufacturer...
On my Insight pump (Accu-Chek), alerts comes from pump, not from pump driver, and pump driver only provide the UI to cancel alert...

This is for me a safety issue but on pump side 🤔

I agree we could probably improve pump driver on this topic to provide an alert from phone, but this will not have the same strength than DASH screamer, and can be disabled within phone settings (I don't know if DASH PDM provides this kind of alert or not...).

vanelsberg, 2024-06-03:
Did you bolus the 10.00U 16:00h before or after the DASH tab was reporting 0.0 insuline left?
And (according to your IOB) this bolus executed successfully?

Note that the fact reservoir is reporting 0.00E left (Pod starts beeping) does not mean that there is the DASH Pod is completely empty. There may still be some units left in the pump. Once a bolus command or set TBR fails due to the Pod being out of insulin, I expect the DASH pump to report a failure or scream?

Philoul, 2024-06-03:
In screenshot I see that latest bolus was delivered 2h22min earlier and was probably a SMB (0.15U).
But we don't see if 10U bolus started, or not (only the dot in main graph).

emmatovar27, 2024-09-02:
In the dash preferences you can set up an alarm for low reservoir.  

The minimum is 5 this can help as a reminder to change the pod! :) 

[image]

closed 2024-11-13 by MilosKozak