nightscout/AndroidAPS #3353: Improve Disconnect Pump Funcionality / Ease of Use

MowgliChild, 2024-05-29:
I believe disconnected pump / no CGM with stick functionality could be improved. 

It would be more helpful if clicking 'disconnect pump', ALSO:

1) Set TBR = 0 (this I believe currently has to be manually enacted). If the pump is off, there can be no basal yet not setting 0TBR affects later IOB calcs since the basal is calculated as delivered. 
2) That the alarm for temporary basal rate error is silenced / switched off. There is no basal as there is no pump. Im not sure why the error is flagged / alarmed?
3)That the alarms are (optionally silenced) for CGM/pump comms. I normally have switched on (in case the pump battery dies, or there is no CGM for some reason) but the current disconnect pump needs. 
4) Consider recording blood stick readings?
5) That pen injections record only are added. This is already an open ticket. #3254 

Background:
I do water sports with pump off. I usually take the phone (and pen) to the beach so I can get CGM readings when resting.. After going out to sea, the TBR error alarm goes off even with the phone volume down (maybe alarm is notification volume?). Setting 0 TBR, alarms off are separate independent actions necessary

Philoul, 2024-06-01:
Can you give us which pump?

1- is included since several years (when you disconnect pump from overview, then a 0 TBR is sent to the pump for selected duration (but you can have within pump some max duration constraints 🤔).
4- has been included in dev branch
2- never got TBR alarms could be linked to pump driver or very long duration (can you give more information on this topic)
3- I personnally don't agree with this request. Buttons are for pump disconnection, not for CGM. (you can tune or disable CGM alerts according to your needs).

MowgliChild, 2024-06-04:
Thanks for your reply. Im unsure about your numbering of the points so will do my best. I dont know how it is 'supposed' to work and the FB post didnt throw any solutions.

Sure, pump is Dana I. 

What more info would you like? The screen dumps are on the linked FB post. 
It says "tempbasal delivery error, connection timed out, Options: a) mute 5 mins b) mute c) ok ".  
It can be quite soon after disconnect, although I regularly hit disconnect for showering and it doesnt sound then. Im happy to do any test you need to clarify.

I understand your reluctance for e.g cgm disconnect. 

However, we dont have a human disconnection button, but it is the functionality Im suggesting here for people leaving their phone AND pump behind. 
Its not the same as pump (alone) disconnect which is why I suggested CGM being configurable add on. There are a lot of steps to override as listed. 
Its unclear what the alarm volume is to turn it down. Turning the phone volume down works when it is sounding, but doesnt work when turning down the phone pre-emptively. The alarm still sounds with notification volume switched off. 

I understand the loop doenst work with no CGM but you can enter stick BG in the BC but (I think) its not logged anywhere? Maybe this is a different ticket?

Thanks

Philoul, 2024-06-04:
It's not because pump is disconnected that pump can be out of range of your phone: Pump should remain within phone range (for BT connexion) even if disconnected.
=> The error you got (to my understanding) is not because pump was disconnected from your body, but only because pump remains far from phone for a longer duration than the duration specified within alarm parameters, so when phone tried to update the 0 TBR, you got the alarm. For me it's not an issue. Keep phone and pump close together when disconnected, and it should fix alarm issue linked to the pump.

I definitively do not understand your CGM topic... A CGM sensor cannot be disconnected, it remains on your body and continue to work and record CGM values even if far from your phone (then when phone come back within range history is synchronised again (at least with FSL or Dexcom sensor, the 2 CGM I know...)
=> You can do manual measure of your BG, but if you are close to your phone you should also get CGM values
=> If during your sport practice you are far from both your phone and your pump, then you can disable CGM alarm (but it's not the same alarm and the same topic than pump alarm...

Note concerning facebook link, on my phone I can't see your facebook message, I arrive in the head of AAPS FB group (I had to connect to my computer). So it's always better to put all information (especially screenshots) here directly into the issue and not with a facebook link...

MowgliChild, 2024-06-04:
Thanks for your comments.

Its hard doing this by messages because it could be easily fixed by conversing. Its hard to see what the other person doesnt understand via text and that is where we are now. I appreciate its what we have. Thank you. :)

I leave the pump in the car and take the phone to the beach for the CGM. It the moment I have to take my meter because I cant switch the unnecessary TBR alarm off without switching the phone off.  If someone stole my bag I wont lose my pump (and phone) if its (theyre) in the car. Also it (they) would sit in the hot sun all day for no benefit.  Im pretty sure that anyone doing watersports including swimming would do this or similar. I dont think it is unreasonable that if you disconnect the pump that it stops trying to update TBR, this is the key issue.  What is the purpose that AAPS apps is trying to update TBR of a disconnected pump? 

Im not a coder, but I am an engineer.  I understand how things could or should work, but have no idea how or why in this case they are implemented. Formulating questions is even hard for most users. 

We have a debate in the community of what the CGM actually is. Yes the HW is fixed to my body, but the software is on my phone. You need both parts obviously to be complete. I said it may need a separate ticket. but CGMs go down for all sorts of comms reasons and leave us with sticking. Ive lost a full Anubis rig in the sea. :(  
Ive been there and thought that it could be handled better, like logging stick data and extrapolation between points with data as it already does for cgm gaps, although it wont affect dosing outside of BC. I put it on this ticket as in this circumstance its connected. Sticking with wet hands is hard. But its not as hard as trying to navigate a phone with wet hands. Im not making a case for one or the other, but see situations where both would be useful. 

Id like to see:
Pump disconnect gives zero TBR and also stops trying to update and alarm for TBR (items 1&2 on my original list). This is the main issue. 
Pump disconnect optionally also cancels CGM alarm for disconnect ONLY) and pump disconnect alarms too (point 3). It would be good if the one button disconnect did all. Yes I could go in and switch off the alarms individually of course. Its not so much about the 'effort' but that in automating the disconnect it is more reliable....so when I reconnect, the alarms reliably come back on.  
Pen bolus in pump off (point 4 you have in dev already, great job). This was more affirmation that this is a great idea. 

Thanks.

Philoul, 2024-06-04:
I will let @MilosKozak decide / answer your request, but I'm afraid it will have huge functional impact related to a very specific use case, I will explain why according to me:
- several pumps have safety constraints concerning TBR duration (generally limited to 2 hours max). So AAPS cannot send a TBR request longer than 2 hours.
- Disconnection duration can be up to 3 hours, So we should either be able to update TBR (current behaviour but request but requires pump connection), or limit disconnection duration to 2 hours (you will probably not agree in your use case), or let pump switch bach to basal profile (which is wrong because you will not get insulin provided by the pump but not within your body).

If pump switch back to profile but remains disconnected and out of phone range, you will get an update of delivered Insulin when pump will be back within Phone range and will synchronize history data. This insulin cannot be ignored by AAPS (other use cases can select disconnect pump with patch pump remaining in body), so that will lead to wrong IOB...

And regarding alerts concerning pump disconnection it's a safety feature...

MowgliChild, 2024-06-04:
Thanks for the info re max TBR duration. I didnt know this. ( Q 1 below) I appreciate it would be easier if they are all treated the same.

Philoul, 2024-06-04:
Keep in mind than whatever the user request, no compromise will be done concerning safety on automated requests. Manual request will override automated commands.

Not sure of 120 min as a safety constraint for pumps (max manual TBR can be up to several hours, and after verification pump constraints seems to be between 8 hours and 24 h), but it's a safety max duration for an automated TBR sent from loop. These safety requirement is link to a lost of phone connection while looping (phone lost, broken, battery down to 0, forgot at home,...).
Whatever the command sent by the loop, it should keep you in a safe situation even if you cannot send a cancel counter request in one run later because of phone out of range. Safe situation is profile basal rate (pump user without loop).

Concerning your questions, not 100% sure of everything but I will try some answer.
> Why do I get TBR alarms sometimes within minutes?

When a command is sent to the pump, if connection failed you can get an alarm immediatly (timeout is a few min only). When you disconnect pump, this will prevent loop to send command to the pump (max disconnection duration is 3 hours), after phone will try to connect to pump and raise an alarm if it fails. I don't know how alarm is managed if you have a shorter duration for alarm compared to disconnection duration...

> Why does pump disconnect not override everything?

No compromize concerning safety, all injected insulin recorded within pump history must recorded as injected insulin within your body whatever pump disconnection status. This is to avoid risk of insulin overdosing. (for ex if during a disconnection you connect temporary pump to provide a bolus manually from pump UI, you cannot ignore this bolus because status is disconnected... Some user can use disconnection feature for patch pumps, that by definition cannot be disconnected, to force a zero temp...)

> What is this max TBR request time for Dana I?

Technical constraint is I think 24 hours

> What happens to the ABSIOB maths if I switch off pump (debattery) for 3 (or more) hours?

I think no 0 TBR will be recorded within pump history, so you will probably get standard profile basal rate take into account for abs iob (not 100% sure), so probably an over estimation of abs basal. (that's why a disconnection should be recorded to set a zero temp).

> What would you suggest for a work around? Is there any way to delete non delivered TBR?

Workaround could be to disconnect pump for max duration (3 hours) to record the 0 TBR (or set manually a longer 0 TBR at your own risks), and close immediatly AAPS application (from 3 dots menu). This should prevent alarms.
You can still read BG values from CGM app (xdrip, BYODA...).

You can delete treatments within Treatments (3 dot menu).

@MilosKozak correct me please if I made mistakes in my answers/explains...

MowgliChild, 2024-06-06:
@Philoul Thanks for your generosity in taking the time to explain.  

Isn't pump disconnect a manual command and therefore more 'important' than an auto of re-establishing TBR?

I appreciate the benefit of auto establishing base pump settings generally, but pump disconnect I would view as a master kill switch. If you dont have a pump connected, any insulin AAPS thinks it has delivered for any reason is 'fake'. I cant think of any safety exceptions to this, its a physical fact, unlike lost comms failures which would indeed be better with a standard basal rate applied.

When reconnected I can indeed delete treatments, but TBR isn't a 'treatment'. So if it thinks its delivered basal and hasnt, I cant delete it. I have had to deleted SMBs where they have fired into my glove box.  Im also effectively bolusing for basal. I dont know how this should be logged.  The noise from other variables (stress, exercise etc) is probably more than the effect of programmed basal rate anyway.

3 dot>exit fully shuts down AAPS but it auto restarts, freezing action buttons until it first reloads pump history. 

Pump disconnect should auto trip a 0TBR for the time period selected, and AAPS currently doesn't do this. Manual 0TBR seems an unnecessary additional step that could / should be automated?

I can manually kill TBR for an indefinite period. I do need the phone on for CGM and for manual bolus logging. I was running VP to manually log only boluses, but it doesnt fix the TBR issues although it may kill the alarms....since it can enact virtual TBR changes?

How about switching to VP and then forcing 0TBR for indefinite period for a work around? 

Thanks

Philoul, 2024-06-06:
> If you dont have a pump connected, any insulin AAPS thinks it has delivered for any reason is 'fake'.

This is true for your veŕy specific use case (tube pump physically disconnected) but wrong in several other possible use cases:

1) Users of Patched pump (DASH, EROS, Medtrum), also use "Disconnect pump" to force a zero TBR (for ex during exercise).
- Pump is at a disconnected status but remains connected on body

You cannot ignore insulin injected during this period

2) Other use case with a tube pump:
In the middle of a "disconnected period" you want to deliver a bolus with the pump, or manually resume pump and don't have your phone available (phone battery is off). then you plug your pump and send a direct bolus from pump and/or restart basal manually. You can't ignore this bolus / basal later for obvious safety reason...

> Pump disconnect should auto trip a 0TBR for the time period selected, and AAPS currently doesn't do this.

Wrong, or it could be a specific issue within your config... 
Each time I select pump disconnection, a zero TBR is automatically sent to the pump for selected duration (Insight pump, confirmed within pump screen).
If I manually re-connect the pump before the end of selected duration, then profile basal rate is immediatly and automatically sent to the pump (and rate will be updated if necessary on following loop run).

MowgliChild, 2024-06-06:
If you set disconnect for exercise and 0TBR ensues, under what circumstances would you want any insulin given? -  having manually (top level) told AAPS you dont want any for the time period selected?

Yes a patch is always physically there, but how can disconnect pump mean anything other than no insulin to be given for any reason (until 0TBR) times out)? 

And even then it would be recorded. The user sets the timer, and it would be better to re-initialize basal delivery as you say for no comms but this w/could be handled with loop on.  BUT the user has manually overridden this by 'disconnect pump'. As you say, this is a top level command. 

It IS more important to log delivered insulin than to log fake insulin, this is true.

Why do I need to set 0TBR if I disconnect pump? Is this user (me) error?

Re-starting a TBR because the disconnected pump timed out is an automation, and only applies to patch users securely, for the same reason. I dont think you are following your own logical arguments / rules here.

Manually delivered insulin (by pump) is checked by AAPS anyway? I'm not suggesting it should be ignored. If I cancel a  bolus delivery half way through, it rightly throws an error, and on checking AAPS is aligned with pump history.  Conversely, if I have to shut off the phone so that the alarms arent going crazy, Id have to switch it on, connect to the CGM, switch to VP and then log manual boluses and then switch it all off again. 

You unquestionably have more SW knowledge than me and make some good points, thanks for your interest and time. But there are some points that don't currently make sense (to me) and there is no obvious work around?

I dont think its that small a population. Anyone who wants to run disconnected a while. It could even be forced as you dont have a spare canula / pod with you. You can manually enact a 0TBR for 24hr. I dont really see why its different.

Jackenmen, 2024-06-09:
> * Consider recording blood stick readings?

Technically, there is one case in AAPS where BG readings are already being added as careportal events automatically. That case is the Contour Plus/Next Link glucometer which, when configured, sends the BG level to a Medtronic pump and the readings can then get synced on the next refresh in AAPS. That is a unique situation though where the glucometer already communicates with the pump and it was simply possible to start handling the event that records that value. I believe the Eversense plugin also adds the BG level you input during calibration but I guess that can at best be called semi-manual since I don't believe Eversense app supports syncing with any glucometer.

MowgliChild, 2026-05-14:
Hey family,

So yesterday I got out of the shower and forgot to put my pump on again. I realised when pump dusconnect alarm sounded but already on my way and have pen support.

I could not cancel basal delivery (obviously fails to update) and this carries over to the maths (aaps thinks basal is delivered). Alarms every pump loop

This ticket hasn't moved in a long time. 

I also think a BC with whole/half unit pen would be helpful with pump off. If im eating i have to do the maths manually extracting settings to calculate dose, when AAPS could do this?

Any comments?

Thanks for an awesome job. Looking forward to V4.0

[image]
[image]