nightscout/AndroidAPS #3358: BUG : Bolus interrupted on empty pod / COB lost

robertrub, 2024-05-31:
Tested with Medtrum Nano.

Only 3U are left. I use bolus calculator for 40gr carbs thus 5.2U.

Nano started bolusing without any alerts (not normal)

Bolus stopped at 3U with screaming alerts (normal)

Carbs were NOT added !!! IOB said 3U but COB was at 0 (not normal).

When new pod was activated, bolus didn't continue with the missing units (normal I suppose).

Philoul, 2024-06-01:
If I remember well pump driver should return to bolus calculator a "successful result" on Bolus delivery to record Carbs correctly (even if bolus was only partially delivered).

So here on 5.2U request you should get a "true" result (but with only 3U recorded in database...

What happen if during bolus delivery you cancel delivery to not inject all what was calculated?

MilosKozak, 2024-06-01:
exactly. carbs are recorded only on bolus success to prevent overdosing on next bolus wzard attempt

robertrub, 2024-06-01:
In this case, partial success ;)

As carbs and bolus are not physically connected, I think that the carbs should be added to the COB even if the bolus was not given (foe any reason).

I start eating and use calculator to enter 40g and is says 6U to bolus. Even if the pump is unreachable or has 0U left or ..., I'll eat the 40g. Thus the need to add carbs to COB even before starting to inject (thus, separating carbs and bolus actions).

Philoul, 2024-06-01:
@robertrub if I compare with insight driver, when I send a bolus with not enough insulin. it's ok without alarm (I think it should work like that)
When reservoir is empty then I get alarm from pump (and Carbs + partial bolus is recorded)
Then after reservoir change if bolus is resumed, then bolus amount is updated, if not, then only delivered insulin is recorded.
The most important point is to get the alarm from pump to be aware of the error and let you check evrrything (for ex launch again calculator to calculate missing insulin after patch pump change).

jbr7rr, 2024-06-01:
> exactly. carbs are recorded only on bolus success to prevent overdosing on next bolus wizard attempt

I find this logic not really intuivite, the user already added the carbs? Also this is only when using the bolus wizard not when recording carbs separately ?

What is the desired outcome when only part of the bolus was delivered? (carbs recorded or not?)

I will have a look into how other drivers handle this as well

robertrub, 2024-06-01:
User can either add the carbs first and use the calculator (without carbs of course) or use only the calculator and enter carbs there.
In both cases, the carbs should be added to the COB counter (insulin delivered or not).

In my exemple, I ended with 3U but no carbs. If I hadn't added carbs again, then AAPS would be "lost" with high IOB, no COB and BGs raising...

emilisev, 2024-06-02:
I experienced similar issue when using the calculator and then the connection to the pump (medtrum also) was lost during injection : carbs were not recorded. Bolus was administered in full but AAPS did not get confirmation so it was not recorded. But I found it confusing that carbs were also not recorded in this case.

jbr7rr, 2024-07-13:
@MilosKozak 

This is the code that determines the behavior:
https://github.com/nightscout/AndroidAPS/blob/08ffac78e198635a44391d0bc6b31a65eec9fec2/pump/medtrum/src/main/java/info/nightscout/pump/medtrum/MedtrumPlugin.kt#L395

This is the same for dana:
https://github.com/nightscout/AndroidAPS/blob/08ffac78e198635a44391d0bc6b31a65eec9fec2/pump/danars/src/main/kotlin/app/aaps/pump/danars/DanaRSPlugin.kt#L320

So can we conclude that this is desired behavoir? As seen in this issue, users do get confused by this, but it also makes the user alert and force them to check delivered insulin.

closed 2024-11-13 by MilosKozak