nightscout/AndroidAPS #3367: Timeline in History screen is not Always correct

francoisqueru35, 2024-06-15:
In the history screen, while switching scale, the timeline is not correct for 6 and 12 hours.
For 6 : the current hour appears two times (see screenshots attached)
For 12 hours :  the graph is one hour wrong (i.e 7 instead of 8 on the screenshots 

Any idea why?
[image]

[image]
[image]
[image]
[image]
[image]
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

fezulin, 2025-01-02:
I observe the same behavior in both version 3.2.0.4 and version 3.3.0.0.
In the first image: On the left is a part of the "History" (This is not a screenshot from the main view!). On the right is a screenshot of the "Treatments".
1.) Left grey arrow: 
Apparently, two SMBs were delivered around 5:30 AM. 
2.) Right grey arrow: 
According to the "Treatment", the SMBs were given at 6:30 AM. One hour later! (And this is the time I remember.)
3.) Red square: 
The "08" is missing. So, an entire hour is missing. 
The same issue happens with entered carbohydrates (KH) and manual bolus (not visible in this screenshot). The displays in the graph are sometimes shifted by one hour compared to the labels on the axis.
[image]

In the following image: The scale is also incorrect, too. Here, at 8 PM, one hour "too much" is displayed.
[image]

I encounter this behavior repeatedly. I have adjusted the time interval (6, 12, 18, 24 hours) several times, and I have shifted the time window multiple times using the arrow keys. In the process, I accidentally set the display to a time period in the future. Now the behavior is like this. However, I don't know what caused the display to be incorrect in this sense. 
So I don’t know how to reproduce it.

pnharris02, 2025-01-22:
I also experience variations of this behaviour.  I remember it occurring back in Sep/Oct in v3.2.0.4, on a Samsung tablet, and it's still occurring in v3.3.1.3 on a Pixel 9 Pro phone.  My experience is a bit more like @fezulin reports, but I have a different interpretation/understanding of the problem, at least in my case.

I believe it is only the scale on the X-axis that is displayed incorrectly, with all data on the graph otherwise shown correctly.  No data is missing.  I assume that the time periods displayed are meant to start from 0:00 (for 24hrs), 00:00 or 12:00 (for 12 hours) and 0:00, 6:00, 12:00 or 18:00 (for 6 hours).  That appears to be match up with the DATA I see on the graph.

However, the scale is wrong.  
* Any time period starting from 0:00 will often show the first label on the X-axis to be 23. Then at some point later, it may or may not skip an hour, so that the last label on the X-axis is correct.  
* I see this behaviour occurring regardless of the time-frame selected, whether it be 6,12,18 or 24hrs.
* It is easily reproduced by starting at the current day, then skipping back using the arrow at top left.  Some periods display with the correct axis labels, and others don't.  Some will display correctly, some will start and finish with the wrong time on the axis, others will start on the wrong time and finish on the correct time, or vice versa
* If the time period is set to 6 hours, then I also see some periods displaying the same hour twice on the axis

pnharris02, 2025-01-22:
Sorry, I just found that there are previous issues that raise the same point and @MilosKozak has said that it is a bug in the library and therefore unsolvable.
https://github.com/nightscout/AndroidAPS/issues/2854
https://github.com/nightscout/AndroidAPS/issues/3217