nightscout/AndroidAPS #3368: AAPS thinks it's outdated even though latest version is installed - closed loop disabled

idowolf, 2024-06-17:
**The issue:**
When using AAPS for over 30 days, even after updating to the latest version on either master or dev branch (the issue has been reproduced on 3.3.0-dev-b as well as the latest master branch version), closed loop will be disabled and IOB will be limited to 0.0. When checking the logs one log reports "Version Checker: old version".

**The cause:**
On dev branch, VersionCheckerUtilsImpl.kt this function is deifned:

```

    private fun checkVersion() =
        if (isConnected()) {
            Thread {
                try {
                    val definition: String = URL("https://raw.githubusercontent.com/nightscout/AndroidAPS/versions/definition.json").readText()
                    val version: String? = AllowedVersions().findByApi(definition, Build.VERSION.SDK_INT)?.optString("supported")
                    compareWithCurrentVersion(version, config.get().VERSION_NAME)

                    // App expiration
                    var endDate = sp.getLong(rh.gs(app.aaps.core.utils.R.string.key_app_expiration) + "_" + config.get().VERSION_NAME, 0)
                    AllowedVersions().findByVersion(definition, config.get().VERSION_NAME)?.let { expirationJson ->
                        AllowedVersions().endDateToMilliseconds(expirationJson.getString("endDate"))?.let { ed ->
                            endDate = ed + T.days(1).msecs()
                            sp.putLong(rh.gs(app.aaps.core.utils.R.string.key_app_expiration) + "_" + config.get().VERSION_NAME, endDate)
                        }
                    }
                    if (endDate != 0L) onExpireDateDetected(config.get().VERSION_NAME, dateUtil.dateString(endDate))

                } catch (e: IOException) {
                    aapsLogger.error(LTag.CORE, "Github master version check error: $e")
                }
            }.start()
        } else
            aapsLogger.debug(LTag.CORE, "Github master version not checked. No connectivity")
```

The function isConnected() is defined as such:

```
    private fun isConnected(): Boolean = receiverStatusStore.isConnected
```

Where the implementation of ReceiverStatusStore (ReceiverStatusStoreImpl.kt) defines isConnected as such:

```
    override val isConnected: Boolean
        get() = lastNetworkEvent?.wifiConnected ?: false || lastNetworkEvent?.mobileConnected ?: false
```

However, when this function is called on app launch, there is no network event yet, so this getter returns false.

**Potential solution**
I solved the issue manually by triggering checkVersion() later, i.e. after 1.5 seconds, and it solved the problem, however a more permanent solution could be triggering checkVersion() whenever there is a network change or at least the first time the network changes to "connected" during the app's lifecycle.

**Other details**
Recreated on version 3.3.0-dev-b as well as the latest master branch version
My device is Galaxy S23 Ultra

vanelsberg, 2024-07-07:
Never experienced this problem (Galaxy S23 Ultra). Not even on a phone without SIM (WiFi connected only)

aemad37, 2024-08-05:
Just for the record I have the same problem where it's reporting a version mismatch 
I'm on version 3.2.0.4 for both sides - ns15.0.2
Clients: s23 ultra and s21 ultra both are connected sim and wifi
Aaps: Xiaomi mi 11 only wifi no sim

MilosKozak, 2024-08-08:
improved by https://github.com/nightscout/AndroidAPS/commit/b5823eeaa1a1044f71bdc90cc6f82107d0515461

closed 2024-08-08 by MilosKozak