nightscout/AndroidAPS #3372: Loop gets kinda sleep mode.

zibra, 2024-06-21:
What could be the reason that Loop goes sometime into kinda sleep mode, so it doesn't really update basal rate or do SMB. t is also not disabling basal on low BG. AAPS application in general works fine (but in background, phone is locked). Connection with pump (omnipod) is fine. It respond to sms commands and applies them. I can execute bonuses, change basal etc. But OAPS last action time is just growing red. To get it back from this state I need to unlock the phone and reveal AAPS app. Phone: Samsung Galaxy S21. AAPS version 3.2.0.3.

MilosKozak, 2024-06-24:
you must look at main phone, not AAPSClient

zibra, 2024-06-26:
Look for what? It's same there. OAPS is frozen. When I look on main phone it unfreezes. My question is why it freezes sometime when mine phone is locked at night.

olorinmaia, 2024-06-27:
Check battery settings and turn off play protect

closed 2024-07-01 by MilosKozak

zibra, 2024-07-03:
I've double checked all settings. Updated some, but no change. OpenAPS is still hanging. Is there any way to debug it / check the logo to understand why?

olorinmaia, 2024-07-03:
AAPS and AAPSClient both use same version of NSClient?

zibra, 2024-07-03:
What do you mean by NSClient version?
I've built them (AAPS and AAPSClient) together from same commit, so probably yes but cant find NSClient version info.

I've checked the logs from AAPS and noticed that when all is fine, logs from: `OpenAPSSMBPlugin.invoke()` appears every minute / max every 2 minutes.
When OpenAPS is hanged, these log lines are missing.

Other logs (from other plugins like NSClientPlugin or AutomationPlugin) are still appearing every minute when OAPS is frozen, so I don't think it's something with my phone blocking AAPS in general but something with `OpenAPSSMBPlugin`...?