nightscout/AndroidAPS #3373: SMBs not disabled but aren't executed

alnassermahrouqi, 2024-06-22:
SMBs have been disabled for two weeks now, loop wants top give SMBs but doesn't execute them
logs will be sent now

[image]
[image]

MilosKozak, 2024-06-24:
unreadable resolution
reset logging to defaults, and provide new logs

alnassermahrouqi, 2024-06-24:
Duplicate treatments probably caused by incorrect xdrip settings when swapping sensors. Once treatments were deleted, SMBs were back

closed 2024-06-24 by alnassermahrouqi