nightscout/AndroidAPS #3375: Communication failure in Omnipod Dash

mordechi-david, 2024-07-03:
All users of the Blackview n6000 device report an error every few hours that AAPS does not find the pod the official bluetooth app crashes until the device crashes and restarts, the only solution I have found so far is to turn off the bluetooth until the AAPS direct protocol finds the pump with bluetooth off and then I Turns it back on so that the Dexcom will continue to work
I looked for logs and I don't have any, I don't know why the device doesn't create logs
I'm on version 3.3.0 dev b but most users are on version 3.2.0.4 and they too have the exact same phenomenon
I will be very, very happy if someone finds a solution because we don't want to change a device, it finds the pod in one second and its battery is very strong
In short, we are very pleased with the device for AAPS
Thank you
I will try to find logs and upload

Philoul, 2024-07-05:
For log files, you can send them by mail from AAPS > Maintenance > Send log file by mail
-> you can send them to you an then copy them here.
You can also see them from computer (smartphone connected by USB)
Go to Android/data/info.nightscout.aaps/... (not sure of exact folder name).

Log files can be hidden from phone on some Android versions.

gerison77, 2024-07-06:
Hi, I have 2 N6000 rigs running with dash. Can't confirm the reported behaviour. Just lost one pod instantly after activation. Once I had to restart phone to regain connection but since this was a onetimer I blame the pod, not the rig.

vanelsberg, 2024-07-07:
Afaik you can not run AAPS and DASH app concurrently (when the DASH app connects to the Pod it is blocking access for AAPS.
Uninstall the DASH app and try again?

gerison77, 2024-07-07:
Ok, what do you mean by official Bluetooth app? There should be just AAPS connecting to the pod.

gerison77, 2024-08-01:
> off the bluetooth until the AAPS direct protocol finds the pump with bluetooth off

If I understand your sentence right, AAPS can connect to Dash without bluetooth on?

The issue mentioned actually happend right now on a N6000. It reconnected, when I shut down the Diakemapp, but deaktivated bluetooth as well. Can there be a conncetion?

vanelsberg, 2024-08-01:
> Ok, what do you mean by official Bluetooth app?

Actually, I meant the official DASH PDM (which is an app running on an Android PDM device):
You cannot have both AAPS and the DASH PDM connected on the same DASH pod

vanelsberg, 2024-08-01:
> AAPS can connect to Dash without bluetooth on?

Nope. What I think writer meant (or is seeing) that turning BT _off_ and then _on_ again solves his problem.

vanelsberg, 2024-08-01:
> Blackview n6000

We have seen some phone brands or models having BT connection instability problems or even unable to find the DASH Pod at all. As more users are reporting problems for the N6000, most likely it not compatible with DASH.

gerison77, 2024-08-03:
N6000, most likely it not compatible with DASH.

I wouldn't say that at this point. I have two N6000 rigs running and they are working. Since I restarted the diakem G7 app, the issue is gone.

mordechi-david, 2024-08-04:
I meant as @gerison77 wrote that when I turn off the bluetooth it communicates in AAPS internal protocol and then it works fine
Usually it also works with Bluetooth turned on, but sometimes problems suddenly start and it causes the Google Play Service application to crash, then the solution is to turn off Bluetooth

https://github.com/user-attachments/assets/c62e2dc1-aa27-4dbb-a247-d4d21bad9084

mk1950, 2024-08-04:
It doesn't help me, even if I turn off bluetooth for two hours it doesn't work.

gerison77, 2024-08-05:
> I meant as @gerison77 wrote that when I turn off the bluetooth it communicates in AAPS internal protocol and then it works fine Usually it also works with Bluetooth turned on, but sometimes problems suddenly start and it causes the Google Play Service application to crash, then the solution is to turn off Bluetooth
> screen-20240804-112103.mp4

Hello! I as it was stated above, there is no internal protocol. There is only a Bluetooth connection.  Without Bluetooth enabled, connection can't be established to POD.

mk1950, 2024-08-05:
> > It doesn't help me, even if I turn off bluetooth for two hours it doesn't work.
> 
> Can you please state if you are using G7.

Dexcom g6 + Omnipod Dash

mordechi-david, 2024-08-05:
It's a fact that when I turn off the bluetooth I manage to connect to the pod, I can upload more screen videos
Another fact is that the pod does not appear in the list of Bluetooth devices, which means that the pod certainly connects via Bluetooth and requires Bluetooth authorization, but not through the built-in Bluetooth application in Android but directly from the loop, so even when Bluetooth is turned off it works
Unlike Dexcom, as soon as I turn off the Bluetooth, it stops working

gerison77, 2024-08-19:
Having checked now this issue with two N6000 over the last two weeks. I can't confirm, that AAPS is able to establish a connection with bluetooth off.  Actualisation just fails. 

However, I made the observation, that connection can be re-established after a loss, by shutting down Diakem app.

There might be a conflict.

closed 2024-11-13 by MilosKozak