nightscout/AndroidAPS #3378: Invalid pump history data detected.

ddeityy, 2024-07-12:
Sent the logs.

MilosKozak, 2024-07-17:
[AndroidAPS.log](https://github.com/user-attachments/files/16267016/AndroidAPS.log)

MilosKozak, 2024-11-13:
@andyrozman

MilosKozak, 2026-10-15:
Found the cause in the attached log.

At 09:46 the history read got page 0. It had only 5 entries, because the pump had just started a new page. The pump did not answer the request for page 1. AAPS still processed page 0 alone:
- The TBR from 09:21:37 was ended at 09:46:27. The pump really stopped it at 09:29:39, but the entries in between were on page 1.
- The last read entry moved to 09:46:29, and the next read starts 35 min before the last read entry.

At 09:56 a full read found the 09:29 - 09:33 entries as new. They were older than the 09:46 TBR that was already processed. So the 09:46 TBR was paired with the 09:29:39 cancel, which gave a negative length and this notification. The 09:33 TBR was also stored as running until 09:56, but it really ended at 09:46.

If the missing entries had been more than 35 min old (for example after a longer radio outage), they would never have been read again. A bolus given on the pump in that time could then be lost.

Fixed in 50c8e9b40b:
- A history read that stops on an error is not processed at all. The last read entry stays as it is, so the next read asks for the same time again and also gets the missing entries.
- A page request that the pump does not answer is sent again. Before, AAPS sent ACKs for about 70 s for a page that never started.

Closing. If you still see this notification on a build with this fix, please open a new issue and attach the logs.

closed 2026-10-15 by MilosKozak