nightscout/AndroidAPS #3384: Cannot install the generated AAPS.apk file on my Pixel 8a running Android 14

manfred-h, 2024-07-14:
I was following the steps at [wiki.aaps.app](https://androidaps.readthedocs.io/de/latest/Installing-AndroidAPS/building-AAPS.html) to build AAPS.apk version 3.2.0.4 on my Linux box. The whole build process was successfull, but the attempt to install the generated .apk file results in a message "Unsichere App blockiert - Diese App wurde für eine ältere Android-Version entwickelt und enthält nicht die neuesten Datenschutzmaßnahmen" (translated to English: "Unsafe app blocked - This app has been developed for an older Android version und does not contain the newest data safety regulations", or so).

My Android version is 14, security update 2024-07-05

Do other people see similar problems? Would this work on a Windows system (I hope not)?

MilosKozak, 2024-07-15:
You must disable play protect and enable installing apks from "other" sources

closed 2024-07-15 by MilosKozak

manfred-h, 2024-07-15:
> You must disable play protect and enable installing apks from "other" sources

The "disable play protect" thing was the missing bit, now the apk installed on my device. Thanks a lot for that hint!