nightscout/AndroidAPS #3385: Watchface not available

MTR93600, 2024-07-17:
Hi everyone, 

I recieves the galaxy watch ultra, which is on wear os 5, one UI 6.0.
Complication ok 
TILES with the shortcut to bolus or TT, ok
Watchface aaps not visible in the list in wearable app on the phone or on the watch when i look in the wtachface list.

Any idea how to fix it ?

koelewij, 2024-07-18:
Looking forward to possible solutions to test.

Philoul, 2024-07-20:
It seems that with Wear OS 5 usage of WFF is now mandatory.
So we will have to recreate new Watchfaces with this format and how reorganize code to reuse (and share with current existing watchfaces) the maximum of existing code 🤔.
I don'tknow the capability of customization by code (nor how to implement them) with WFF.
So we will probably have to include new libraries (hope we will not have conflicts with previous one), check and test how to build/update/show graph (for bg curves, prediction lines...) show/update imageviews and textviews...

MTR93600, 2024-07-20:
Apparently the file watch_face.xml need a specific code to make the custom watchface available. The manifest.xml required probably modification too. Then i'm not sure everything need to be rewrite. The change is related to security.

Philoul, 2024-08-01:
We are not alone with this problem...
Facer that have a catalog of more than 500000 watchfaces available within playstore, and millioms of customers has the same issue... See https://community.facer.io/t/important-samsung-galaxy-watch7-ultra-compatibility-update/93887

I understood that if you have a watch with Wear OS 4 with our watchfaces installed, if you manage a full backup of your watch, update your watch to wear OS 5.0 and restore your previous backup after the update, then it can still work (but not sure).

If you buy one of the latest galaxy watch like ultra with Wear OS 5, no way it's locked.

reijoahola-nightscout, 2024-08-01:
I had a backip for my Galaxy Watch 4 (Wear OS 4) in Google cloud and when installed the new Galaxy Watch Ultra, it installed backup from Google cloud. Such apps like G-Watch Wear, xDrip+ was successfully installed from backup and offered correctly complications that also worked in other watchfaces. But AAPS behaved differently. It was clearly copied from the backup and offered complications but they did not work. I could select also AAPS Tiules but I am not sure could they work correctly. I then uninstalled AAPS from the watch and installed it again by using Wear Installer 2. The situation was the same, complications could be selected but did not work. In all these these test, AAPS real watchfaces could not be selected and displayed.

Philoul, 2024-08-01:
@reijoahola-nightscout culprit is google here complain directly to google ( I afraid end user satisfaction is not their priority, but the more users complain, the more we could have a (very little) chance they will take into account our request..  (let the end user choose if they want optimized new watchfaces designed with WFF format, but with less feature, or still use their old watchfaces...)

olorinmaia, 2026-09-19:
@MilosKozak @MTR93600 this issue can be closed due to the merging of #5123 in dev (AAPS 4.0.0)

olorinmaia, 2026-09-23:
Closing issue due to the merging of https://github.com/nightscout/AndroidAPS/pull/5123 in dev (AAPS 4.0.0)

closed 2026-09-23 by olorinmaia