nightscout/AndroidAPS #3391: [documentation] Make list of available AAPS pumps in each country

andyrozman, 2024-08-02:
I would recommend that we create a list of AAPS-able pumps that are available (still being prescribed or supplies still being prescribed) in each country. This might be helpful for new people that want to have this information. 

Maybe with columns: Country, Pump, Supplies, Comments, Last Updated.

We could maybe add special section in Discord, where people could report this data.

psonnera, 2024-12-21:
@andyrozman can you add this to https://github.com/openaps/AndroidAPSdocs/issues?

koelewij, 2025-01-29:
In my opinion, it would become very hard to make this list complete for all countries and especially keeping it up to date. 
It requires constant monitoring of what is available in each country. 
Any idea who should maintain this information and how much effort is required to keep this up to date?

There is nothing worse than people making decisions based on the wrong/outdated information. 
Sometimes insurances and sometimes suppliers decide what goes on and of the market. 

Personally I think its good to advise the potential users to check with their supplier/insurance first before they decide on a pump.

andyrozman, 2025-01-29:
It would be hard to mantain, but it would be very informational. I would also add date when information was last updated and maybe contact person (not email, maybe facebook name or discord name). Easiest would be if this would be just Google Spreadsheet or something like that, so that it can be updated very fast... And we would o corse write that list is just of infromational value and that it might have outdated data on it...