nightscout/AndroidAPS #3394: [Feature request] add predefined profiles modifications

gregorybel, 2024-08-03:
I do profiles modifications several times a day, especially before lunch I increase the percentage for 30 or 60min.

As I do it really often with always the same parameters, I thought we could add predefined profiles modifications like we did it for the temporary target, we could add 2 or 3 default profiles modifications into the profile UI.

The default values can be set using configuration menu and are stored permanently.

We could maybe set custom icons but this would optional.

mick711, 2024-08-03:
Why you don't use automation ? You çan add buttons..

gregorybel, 2024-08-03:
Good point, because I set profiles above 130%

mick711, 2024-08-03:
Oh yes sorry i don't see this hard limit .
You can change hard limite set to 130.0 to your max limit ;

gregorybel, 2024-08-03:
Yes that's might be the easiest solution, how do I do that? I don't see a
parameter for this, shoo I modify directly into the code?

mick711, 2024-08-03:
You can find with find in files : const val MAX = 130.0
File InputPercent.kt

closed 2024-08-04 by MilosKozak