nightscout/AndroidAPS #3401: BUG temp basal increases and SMB in Low Glucoe Suspend

cygrahampearson, 2024-08-07:
[AndroidAPS._2024-08-05_20-00-14_.0.zip](https://github.com/user-attachments/files/16534231/AndroidAPS._2024-08-05_20-00-14_.0.zip)

- occured on Wednesday, August 7th. LGS was enabled via SMS commands. Confirmed enabled by SMS and by the LGS symbol on the device. SMBs were given at 2:47 and 2:57. Temp basal increases also occured
- AAPS [3.2.0.4](http://3.2.0.4/)
Build: 3df83535f9-2024.07.07
- 
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
-

Philoul, 2024-08-08:
Can you please share a screenshot of main overview (including main graph and a subgraph with IOB / COB curves) please ?

cygrahampearson, 2024-08-08:
Here is a nightscout report with COB, IOB and prediction curves. I also attached some screenshots of SMS.

There was a pump unreachable SMS an hour before the SMBs. I wonder if it reverted to SMB rather than open loop somehow?

[image]
[image]
[image]

Philoul, 2024-08-09:
Read the doc concerning ĹGS mode here:
https://androidaps.readthedocs.io/en/latest/Configuration/Config-Builder.html#low-glucose-suspend-lgs

IOB was negative during all the night so LGS worked as designed and didn't provided more insulin that what was defined in profile.
SMBs and High TBRs kept IOB negative.

cygrahampearson, 2024-08-10:
Thank you for clearing that up @Philoul 

I feel like a fool for not knowing that.

closed 2024-08-17 by MilosKozak