nightscout/AndroidAPS #3403: Medtrum patch pump alarms cannot be acknowledged

Sven-Milinski, 2024-08-09:
When using the Medtrum PDM, the user manual states that if a pump alarm occurs, the alarm must be acknowledged in the PDM. If the alarm is not acknowledged in the PDM within 10 minutes, the patch will sound a siren until the alarm is acknowledged.

AAPS currently lacks the ability to acknowledge these alarms. This means, for example, that when the reservoir is empty, a siren sound is emitted that the user cannot stop in AAPS. The alarm can only be stopped by removing the patch and disconnecting the pump base. If no new patch is connected within 10 minutes, another alarm is triggered. Since this cannot be acknowledged, a siren sound is emitted until a new patch is connected. This can be very unpleasant if you do not have a new patch to hand.

I only have a German user manual for Medtrum pump. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

See last row in alarm04.jpg. It's about empty reservoir. There it says 'Tap to acknowledge'. See alarm01.jpg, there it says if an alarm is not acknowledged within 10 minutes patch will emit a siren sound until confirmation of alarm.

In AAPS, a way to acknowledge alarms is needed to achieve the same behavior as with the PDM.

I1985, 2024-08-18:
I've wondered about this as well, but haven't gotten ribs to check the Medtrum documentation (something about ADD and "should do that later"). But I very much support the request to integrate this in AAPS if possible!

jbr7rr, 2024-09-16:
So far I have only been able to test alarm functionality with the Medtrum app. and there is no option to acknowlegde the critical alarms and prevent the base from beeping after 10 mins. Is there someone who is able to test this on PDM and describe the behavoir for that?

closed 2024-11-01 by MilosKozak

blaqone, 2025-03-14:
@MilosKozak Why closed as completed?!