nightscout/AndroidAPS #3404: Insulin or carbs not being sent to Nightscout using Bolus Calculator

Troplo, 2024-08-15:
This is a persistent issue I'm experiencing with AndroidAPS on both NSClientV1 and NSClientV3, with the former only sending the bolus insulin, and the latter only sending carbs. [AndroidAPS.log](https://github.com/user-attachments/files/16624490/AndroidAPS.log) - This log file is a snippet from when I calculated a bolus of 50g carbs and 9.60U insulin bolus using the Bolus Calculator at 20:57, where only the 50g carbs appeared on Nightscout.
[image]
[image]

I've temporarily worked around this problem by emitting the treatments to xDrip+ which then sends to Nightscout, the carbs are duplicated within xDrip+ and it's not a perfect solution as in the documentation I believe it's mentioned that you should disable xDrip+ Nightscout uploading.

AAPS Build:
  Version: 3.2.0.4
  Build: 3df83535f9-2024.05.29
  Flavor: fullrelease
  Nightscout version: 15.0.2

If you need any extra information or I'm simply missing something, please let me know. I've enabled the treatment upload option for the Nightscout client within the AAPS settings.

I'm unsure if this is a Nightscout or AAPS issue either, it appears to send the WebSocket event from what I can see in the logs, it just doesn't appear. xDrip+ has had zero issues in the past emitting carbs and insulin and having them recorded in NS.

MilosKozak, 2024-11-13:
can you re-test with 3.3 dev?
this is not happening to me

Troplo, 2024-11-13:
> can you re-test with 3.3 dev? this is not happening to me

I'm going to close this issue since I can't reproduce it either now, it stopped happening about a month ago when I decided to disable the xDrip+ uploading due to an unrelated issue (I originally enabled xDrip+ uploading to circumvent this issue). I have absolutely zero idea what changed, I don't think I updated NS and definitely haven't updated AAPS since.

closed 2024-11-13 by Troplo