nightscout/AndroidAPS #3406: Feature request (for discussion) add negative Carbs input

Philoul, 2024-08-17:
Use case: sometimes (due to exercise, type of carbs during meal...) Carbs absorption can be much faster than calculated COB with deviation.
So we can have for a long duration zero deviation, a lot of remaining COB and slow carbs absorption calculated by min_5m_carbimpact.
This can lead to over correction on BG variation due to an overestimation of remaining COB.
In parallel lots of users don't enter carbs during exercise (again to not take the risk of over correction with loop).
The idea is to allow negative carbs entry (max allowed value is remaing COB), to reduce remaining COB.
=> for discussion...

swissalpine, 2024-08-18:
I very much support this idea! I discussed this with Adrian years ago. However, my ideas were more far-reaching: enter the impact of movement as a negative COB and enable negative COB ... He saw a big problem with the latter because COBs are seen as 0 or positive values in all calculations. This would not happen with your proposal. 
However, a negative eCarb entry, limited to 0 COB, would also be good.

jbfeutz, 2024-08-22:
Also very helpful for children who may not finish an entire meal. The process to delete a carb entry is very complicated, especially if you are only removing a portion of an announced carb entry.

gerison77, 2024-08-29:
With respect to sports, I would encurage to implent a "fake carb" option as well like the equivalent for "record a bolus only".

Especially during sports carbs are consumed and not entered. However it would be nice to track the intake of the carbs during sports just be able to analyse the amount of carbs actually needed in that particular situation. I have two active TD1 kids. And sometimes I loose track whom i gave which amount when...

MilosKozak, 2024-08-29:
add it as a note

olorinmaia, 2024-09-14:
9fe441bc I guess this can be closed, nice feature :)

closed 2024-10-16 by Philoul