nightscout/AndroidAPS #3408: [DEV]OpenAPSSMBPlugin.kt TDD calculation

szantos, 2024-08-21:
Possibly missing brackets:
[image]

In line 324 there are brackets:
[image]

Nothing to do with the brackets above, but the formulae (with its constants) appear at a 3rd place, too. Would it make sense to define only at one place:
[image]

MilosKozak, 2024-08-21:
missing brackets is definitely a bug ..... I'll fix it

MilosKozak, 2024-08-21:
fixed by https://github.com/nightscout/AndroidAPS/commit/dd40be11cd5d404c3a2c5787601c4f5e7f92d7bf
good catch @szantos !!