nightscout/AndroidAPS #3412: Single treatment not showing up in AAPSClient

jbfeutz, 2024-08-22:
The bolus calculator was used on my son's phone at 10:55 local time. My own phone running AAPSClient was in an area with poor cell coverage at the time, but never received the backfilled treatment when connection was restored.

NSV3 on both devices
Web sockets enabled
Treatment did show up in Nightscout

I run a personal fork of AutoISF, but I don't believe that is related to this issue. If that prevents the issue from being investigated though I understand.  AAPSClient 3.2.0.4+autoISF3.0.1, Build aef96efa0c-2024.06.19

Log file attached
[AndroidAPS._2024-08-21_19-00-00_.790.zip](https://github.com/user-attachments/files/16719004/AndroidAPS._2024-08-21_19-00-00_.790.zip)

MilosKozak, 2024-08-24:
is it visible in NS?

jbfeutz, 2024-08-25:
Yes, it was visible in NS.

jbfeutz, 2024-08-25:
This happened again, this time on my wife's phone. The only thing that has changed in our setup is our son has returned to school. He uses a combination of their WiFi and the cellular plan on his phone. But since the readings are showing up on NS I don't know how this would be causing the issue. Maybe it has been an issue and we're just realizing now because he is away from us for longer periods of time.

MilosKozak, 2024-11-30:
improved here https://github.com/nightscout/AndroidAPS/commit/ed37ffac7effecbe60dd2faeb0bcbcf082d5141c

closed 2024-11-30 by MilosKozak