nightscout/AndroidAPS #3413: Dev-b AutoISF / Can't calculate acceleration

robertrub, 2024-08-24:
I switched G6 transmitters which resulted in a desynchronisation of the 5 minutes interval which started the recalculated values.

[image]

I think (just a guess) that AutoISF plugin does not like recalculated values and can't get a "reliable" acceleration; all acceleration numbers are at 0. 

As it didn't see the deceleration, it selected an agressive ISF for those 3 last BGs (were over target).

[image]

As you can see, just before the drop, there was a dark dot.

Here are the logs (around 22h13 and before)
[image]

ga-zelle, 2024-08-24:
There are only 4 loop events in a single log file. It would help if under Maintenance you disable WEAR (because it just jams the file and makes reading very hard) but enable GLUCOSE which I think is the important one here.

robertrub, 2024-08-25:
Ok, did the changes to wear and glucose last night. Here are the new logs. There are still dark green dots but I don't know if acceleration was at 0 or not.

[image]

[image]

robertrub, 2024-08-25:
Calculated BGs (transmitter change 24h ago)

[image]

Acceleration at 0

[image]

robertrub, 2024-08-25:
OK, I found 2 cases in the logs where Parabola settings change to 0 after a BG time recalculation and the BGs go into a "loop" with 3 minAGO  coming over and over. In the first ZIP, it is 30, 40 and 45 min ago. Look before 9:16:34:840
[AndroidAPS._2024-08-25_00-00-28_.93[1].zip](https://github.com/user-attachments/files/16740609/AndroidAPS._2024-08-25_00-00-28_.93.1.zip)

robertrub, 2024-08-25:
Recalculated data period is over. There are no more dark green dots on the graph and AutoISF is working normally once again.

There is a problem with the recalculated data and AutoISF.

ga-zelle, 2024-08-25:
The problem is with gaps in data, not recalculated ones. AutoISF deals with them as expected.
Without G6 I have recalculated all year round and no such problems..

robertrub, 2024-08-26:
Sorry but I disagree. As soon as the yellow triangle disappeared (and the dark green dots too), the problem of parabola at 0 stopped. Field test says that I'm right ;)

The recalculated data of Libre and G6 seems to be different. No idea what difference. Maybe the difference is that Libre has just _seconds_ of difference when a transmitter change makes _minutes_ of difference...

Another possibility : Are you using the AutoISF plugin in _AAPS dev_ or are you using your "own" AutoISF build ?

robertrub, 2024-11-13:
@MilosKozak Same as the _color changing dots_ issue, please reopen this one too. Both are related and both are not solved.

vanelsberg, 2024-12-01:
https://github.com/nightscout/AndroidAPS/issues/3415

closed 2024-12-11 by robertrub