nightscout/AndroidAPS #3414: Trend arrow in Wear watchface does not account for smoothing

lucasvreis, 2024-08-25:
The trend arrow in the Wear watchfaces apparently do not account for smoothing. See the two attached images below. I've tested it both stock and personalized watchfaces and both have the issue. Collector is a Libre2Plus (CGM Brazilian version) with updates every 1 minute, sent by Juggluco with EverSense interface and received via the MM640g source.

Version: AAPS 3.2.0.1 (aeed6e2083-2023.10.24)

I think logs would not help, but please correct me if they are necessary.

[image]
[image]

MilosKozak, 2024-11-30:
@Philoul still valid for dev?

Philoul, 2024-12-01:
Never got this issue on my side (neither with previous master nor in dev), but probably because I use G6 without filtering...
I made a commit to send to watch the same Arrow calculated in Overview, so it should fix this issue ;-)

soficshift, 2024-12-02:
Thanks so much! Sorry for not providing more details.