nightscout/AndroidAPS #3415: Dark and bright green dots change..

robertrub, 2024-08-25:
I have changed transmitters and got the yellow triangle "recalculated data". Ok, I'm used to it.
The strange thing is that in the graphs, the dark and bright green dots change status/color often... How come ?
[image]

[image]

[image]

[image]

robertrub, 2024-08-25:
[image]

They are all bright now...

robertrub, 2024-08-25:
And dark dots are back...
[image]

MilosKozak, 2024-11-13:
caused by delayed data

robertrub, 2024-11-13:
@MilosKozak You closed this issue but the dots changing color was NOT due to missed readings. I had added the _missed readings arriving later_ info thinking it might help to give you a search direction. All my screenshots were done when the phone was near the G6 and xDrip was receiving the BGs regularly.

I think the problem comes from the smoothing in AAPS but I'm not sure.

I'm about to swap sensors. Maybe the desynchronisation of the 5 min interval between the 2 transmitters is creating a problem in the smoothing section... (very wild guess).

I'll stop smoothing before the swap and see if anything behaves differently...

Please reopen this issue as it is not solved.

robertrub, 2024-11-15:
Stopping smoothing in AAPS does NOT stop the dot problem..The end of 36 hours after transmitter change stops the problem.

So, the "culprit" is with/around the recalculated data code (when there is a "slide" of the 5 min delays when the G6 transmitters are changes).

vanelsberg, 2024-12-01:
> the dark and bright green dots change status/color often...
Hmmm. Using BYODA: do not see these different colors at all. You are using xDrip+?

robertrub, 2024-12-01:
Yes, xDrip.

vanelsberg, 2024-12-01:
Haven't used xDrip+ other then in follower mode. Could it be xDrip+ does some filling in (or Libre smoothing)?
Just asking....

robertrub, 2024-12-01:
I saw once that, if missed 2 readings (far from phone), then on next connection, only reading 3 arrived and 5 min later, readings 1,2 and 4 arrived.

This said, the dark/light green dots (and they change color !!) happens even when I'm near the phone and all the BGs arrive every 5 min.

Look at the 2nd logs I posted on this issue. They are strange...

https://github.com/nightscout/AndroidAPS/issues/3413

MilosKozak, 2024-12-05:
resolved in https://github.com/nightscout/AndroidAPS/issues/3413

closed 2024-12-05 by MilosKozak