nightscout/AndroidAPS #3428: AAPS 3.2.0.4 and NSCLIENTv3 has 10 Entries in queue

TeufelHH, 2024-09-10:
A few days ago I noticed that I constantly have entries in the queue from the NSCLIENTV3 tab. The day before yesterday there were 9 and since yesterday there have been 10. I have already tried a complete synchronization but then when processing it it counts down from around 50,000 down to 10 and what remains. Otherwise everything looks good, I didn't find any errors, my data is apparently all being uploaded to Nightscout.

AAPS 3.2.0.4
NSCLIENTV3
NS-Host: 10be.de
NS-Version: 15.0.2

Last Logfile see attachmend

[image]

[AndroidAPS.log](https://github.com/user-attachments/files/16941921/AndroidAPS.log)

MilosKozak, 2024-09-10:
reset logging to defaults, perform sync and upload logs again

MilosKozak, 2024-09-10:
the reason is not visible. probably not a worth to dig into it
thanks anyway

closed 2024-09-10 by MilosKozak

Gohtraw, 2024-09-17:
Actually, I have the same problem.
I think something went wrong when we deleted an entry from the food database (maybe in NS and at the same time in AAPS).
AAPS tries synching the deletion with NS, doesn't find a record with that identifier and throws a java exception:
```
12:53:45.844 [RxCachedThreadScheduler-44] D/NSCLIENT: [NSClientV3Plugin$onStart$9.accept():241]: ► UPDATE food Sent PairFood 66aa3d0b91785261081fab91 <i>{"carbs":10,"category":"Süßigkeiten","date":1725965625843,"energy":376,"fat":5,"gi":2,"identifier":"66aa3d0b91785261081fab91","isReadOnly":false,"isValid":false,"name":"Milka - Alpenmilch","portion":4.0,"protein":1,"subCategory":"Schokolade","unit":"pcs"}</i> 2/12
12:53:46.151 [DefaultDispatcher-worker-5] E/NSCLIENT: [NSClientV3Plugin.dbOperationFood():573]: Upload exception
java.lang.IllegalArgumentException: URL "v3/food" does not contain "{identifier}". (parameter #1)
    for method NightscoutRemoteService.deleteFood
	at retrofit2.Utils.methodError(Utils.java:54)
	at retrofit2.Utils.methodError(Utils.java:43)
	at retrofit2.Utils.parameterError(Utils.java:64)
	at retrofit2.RequestFactory$Builder.validatePathName(RequestFactory.java:829)
	at retrofit2.RequestFactory$Builder.parseParameterAnnotation(RequestFactory.java:416)
	at retrofit2.RequestFactory$Builder.parseParameter(RequestFactory.java:325)
	at retrofit2.RequestFactory$Builder.build(RequestFactory.java:206)
	at retrofit2.RequestFactory.parseAnnotations(RequestFactory.java:67)
	at retrofit2.ServiceMethod.parseAnnotations(ServiceMethod.java:26)
	at retrofit2.Retrofit.loadServiceMethod(Retrofit.java:202)
	at retrofit2.Retrofit$1.invoke(Retrofit.java:160)
	at java.lang.reflect.Proxy.invoke(Proxy.java:1006)
	at $Proxy5.deleteFood(Unknown Source)
	at app.aaps.core.nssdk.NSAndroidClientImpl$updateFood$2.invokeSuspend(NSAndroidClientImpl.kt:418)
	at app.aaps.core.nssdk.NSAndroidClientImpl$updateFood$2.invoke(Unknown Source:8)
	at app.aaps.core.nssdk.NSAndroidClientImpl$updateFood$2.invoke(Unknown Source:2)
	at app.aaps.core.nssdk.NSAndroidClientImpl$callWrapper$2$1.invokeSuspend(NSAndroidClientImpl.kt:497)
	at app.aaps.core.nssdk.NSAndroidClientImpl$callWrapper$2$1.invoke(Unknown Source:8)
	at app.aaps.core.nssdk.NSAndroidClientImpl$callWrapper$2$1.invoke(Unknown Source:2)
	at app.aaps.core.nssdk.utils.CoroutineUtilsKt.retry(CoroutineUtils.kt:21)
	at app.aaps.core.nssdk.utils.CoroutineUtilsKt$retry$1.invokeSuspend(Unknown Source:17)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:115)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:103)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:793)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
```