nightscout/AndroidAPS #3430: [Bug] Dev/Master Temp Profile switch with duration, reduced % and Activity TT doesn't activate properly on AAPS (Only NSClientV1)

olorinmaia, 2024-09-10:
AAPS 3.3.0-dev-b / master 3.1.0.4
Build: 18887e78ed-2024.09.09
Nightscout version: 15.0.2

Logs:
AAPS: [AndroidAPS_LOG_1726002964538.log.zip](https://github.com/user-attachments/files/16955082/AndroidAPS_LOG_1726002964538.log.zip)
AAPSClient: [AndroidAPS_LOG_1726003014356.log.zip](https://github.com/user-attachments/files/16955090/AndroidAPS_LOG_1726003014356.log.zip)

**Step by step**
1) Select profile switch on AAPSClient phone.
2) Select e.g. 30 min duration, 80% strength, Activity TT and click ok.

**Result (dev):** TT is set on AAPS and AAPSClient, but temp profile switch isn't carried out on AAPS.
**Result (master):** TT is NOT set on AAPS, but is set on AAPSClient, temp profile switch is successfully set on AAPS

**Expected result** is that TT, profile% and duration is set correctly on AAPS.

**Side note**: In dev this works fine with NSClientV3. I haven't tested with master but I assume same behaviour. Since behaviour is not the same in master and dev, it's probably best to fix the problem as it stands in dev if it cant be adressed in both environments some how.

Some screens:
AAPSClient:
[image]
[image]
[image]

AAPS:
[image]

MilosKozak, 2024-11-01:
hopefully resolved by https://github.com/nightscout/AndroidAPS/commit/0ee8b101370d16b58b594aa4794cdfde47e7e7b7

olorinmaia, 2024-11-01:
Unfortunately it still doesn't work in NSClientV1. It works if I first do a temp profile switch then TT afterwards.
When I combine the two actions in one operation the profile switch fails. It still works in NSClientV3 after https://github.com/nightscout/AndroidAPS/commit/0ee8b101370d16b58b594aa4794cdfde47e7e7b7

MilosKozak, 2024-11-03:
Hmmm NS deduping was still blocking records
now it works for me well https://github.com/nightscout/AndroidAPS/commit/11da18ea85579cf3626d8f5539d6d02d4410cfaf

olorinmaia, 2024-11-03:
Thanks for looking into this Milos. It works when you do it from AAPS. But not when from AAPSClient. I tried with both Dash and Virtual pump. After the two latest commits regarding this, I notice nothing happens when I try to cancel Activity TT on AAPSClient after it is set in the same operation as temp profile switch. I have to cancel it one more time for TT to get cancelled in AAPSClient/AAPS.

This is a screenshot of an example I execute on my AAPSClient phone, that doesn't sync profile switch to AAPS.
[image]

MilosKozak, 2024-11-04:
increased to 10s
now working from AAPS client (at least for me)

closed 2024-11-04 by MilosKozak