nightscout/AndroidAPS #3433: Feature Request: Bolus Calculater in AAPS Client, Record Carbs and Calculation only

gerison77, 2024-09-12:
Case: 
Several Caretakers (incl. parents) with AAPS Client in a  Master/Client setup (parental setup) for kids in school. 
Bolus is administered via SMS.

Current situation (AAPS 3.2.0.4): 
Meal Bolus is calculated via Bolus Calculator on client. Bolus is administered via SMS. Carbs are recorded manually. Calculation is not recorded (and therefore can't be checked later on).

Problem to solve: 
Being able to check calculation in order to improve settings. 

Solution and feature request:
New feature in Bolus Calculator: Record Carbs and Calc only, skip Bolus. Which would be administered via SMS.

Ps: This feature would come in handy as well, if Pump is disconnected an Insulin is administered via Pen. I as I learned, the feature to record boli administered by pen with disconnected pump is added already in the current dev.

szantos, 2024-09-16:
IMHO basically a good idea, but only the calculation should be recorded, without Carbs for safety reasons to avoid overdosing.

If Carbs get recorded, it immediately triggers the algo to recalculate and it'll find there is COB, but not enough IOB - therefore it'll dose insulin. Then the SMS command arrives with the originally calculated insulin which gets dosed, too. In total there will be more IOB as should be.

gerison77, 2024-09-16:
Good Point, but to keep things simple I would encounter this issue with a fixed delayed entry of carbs, say 2 to 5 min.

closed 2026-09-09 by MilosKozak