nightscout/AndroidAPS #3435: AAPS client Sync button does not work 3.3.0-dev-c

ohjay73, 2024-09-14:
Unable to access the syncronization settings for appsclient to adjust or view current sync settings. 

You can find the option via a search of preferences, but unable to "click" and edit

build 922640eb63

ohjay73, 2024-09-14:
AAPS full app works just fine

olorinmaia, 2024-09-14:
This has never been possible in AAPSClient if you are talking about the NSclient > Sync option available in AAPS.

olorinmaia, 2024-09-14:
@MilosKozak this is actually a bug, in current master 3.1.0..4 you can search for "sync" in settings and open NSClient Synchronization options that shouldn't be possible. These settings shouldn't be altered and will cause issues if changed on AAPSClient side. In dev, you can search for it and it appears, but you can't click on it. It shouldn't appear in search either.

See screens from AAPSClient:
Option is hidden if you navigate to settings:
[image]

If you search for it, it appears and you can click on it and adjust settings.
[image]
[image]

MilosKozak, 2024-11-01:
in current dev you can search for it but you cannot select "Synchronization"
Can you confirm?

olorinmaia, 2024-11-01:
Yes, it appears briefly, you can click on it, but nothing happens. After a few seconds it disappears.

closed 2024-11-13 by MilosKozak