nightscout/AndroidAPS #3436: Negative carb entry doesn't sync between AAPSClient / AAPS

olorinmaia, 2024-09-14:
Just tested 9fe441bc - nice new feature!

When entering negative carbs in AAPS it doesn't sync to AAPSClient, neither the other way.

Tested with both v1 and v3 NSClient

vanelsberg, 2024-09-18:
> When entering negative carbs in AAPS it doesn't sync to AAPSClient, neither the other way.

Depends on the objective of entering negative carbs.

- If it is used to prevent overdosing due to the fact that according to oRef calculations carbs correctly entered absorbe to slow I think the negative carbs should not be counted or sent to NS/appsclient/other. Carbs were actually taken!
- _If it is to modify carbs entered wrongly earlier, I do not think one should use negative Carbs for this_.

vanelsberg, 2024-09-18:
> It is very important that COB on AAPSClient and AAPS is synced. I cant see any reason why it shouldnt.

Ok. Yes. Then this is a bug...

Didn't think about remoting by parent (which obviously aapsclient is used for)

olorinmaia, 2024-11-01:
Fixed by c12639b7

closed 2024-11-01 by olorinmaia