nightscout/AndroidAPS #3442: Profile switch with Dash sets 4800 min TBR 0 + suspend that doesn't sync well with AAPSClient

olorinmaia, 2024-09-26:
AAPSClient / AAPS 3.2.0.4
NSCLIENTV1
Build: 036071563a-2024.09.13
Nightscout-versjon: 15.0.2

@MilosKozak  @avereha 
Reproduced problem several Dash-users are having getting wrong IOB in AAPSClient because of sync issues. Creating issue to see if we can find a solution. One of the issues reported at facebook: https://www.facebook.com/groups/AndroidAPSUsers/posts/3917923875095667/

Logs AAPS: [AndroidAPS_LOG_1727295509569.log.zip](https://github.com/user-attachments/files/17155205/AndroidAPS_LOG_1727295509569.log.zip)
Logs AAPSClient: [AndroidAPS_LOG_1727295563688.log.zip](https://github.com/user-attachments/files/17155204/AndroidAPS_LOG_1727295563688.log.zip)

When AAPS performs a profile switch (or temp profile switch) Dash / AAPS sets a 0 TBR suspend for 4800 mins (80 hours aka Pod max lifetime), then immediately after set profile basal rate. I think the duration or perhaps how quickly this happens may be the culprit some how to the iob sync issues. AAPSClient doesn't sync this correctly always. Max tbr duration for dash is 720 min if you set a manual tbr.
I wonder why the duration is set so long. And if it really is needed if this is queued? 

Unless a new TBR is sent after Profile Switch AAPSClient will run with 0 TBR while AAPS continues with regular basal rate, this accumulates wrong IOB in AAPSClient. This problem is most noticeable in open loop or LGS mode.

Profile switch I reproduced the error with:
[image]

AAPS:

| AAPS | AAPSClient |
|--------|--------|
| [image] | [image] |
| [image] | [image] |
| Cell | Cell |

vanelsberg, 2024-09-28:
I think 🤔  the logic is that on profile switch a 0 TBR is set indefinitely ("4800 - max Pod live"). It assumes the new TBR will be set once/if the new profile becomes active and on 0Ref calculations are executed on the next BG received. I cannot see any problems in that?

So afaikt most likely this is a sync problem with NS, AAPSclient or both?

olorinmaia, 2024-11-04:
Steps to reproduce: (Easiest in Open loop or LGS, but also happens in closed loop, but on next loop run if TBR is set new TBR is synced to AAPSClient)
* Set 5 min temp profile switch of 120%
* Let it run the duration and let it automatically switch over to normal profile.
* When AAPS reverts to normal profile 0 tbr 4800 min is set then regular basal rate. The regular basal rate doesn't get synced to AAPSClient, but the next TBR does. When it does it doesn't update the wrong 0 tbr that has ran since profile switch causing wrong IOB in AAPSClient compared to AAPS.

MilosKozak, 2024-12-04:
is this valid for v3 too?

olorinmaia, 2024-12-04:
My best guess this is not happening on v3 but can test it when I come home from business trip end of week if needed.

olorinmaia, 2024-12-10:
This doesn't happen with v3. But confirming it still happen with v1. Could a possible fix be to delay the time of setting basal rate after the 4800 min 0 tbr suspend? Something similar to what was done here? https://github.com/nightscout/AndroidAPS/issues/3430