nightscout/AndroidAPS #3443: Simplify-UI is not working on Galaxy Watch 6 Classic.

TeufelHH, 2024-09-27:
Simplify-UI is not working on Galaxy Watch 6 Classic. 

Tested with all AAPS watch faces. 

When i go sleeping, i switch my watch into sleeping mode, the display entering in "grey-mode" but it doesn't show me the simplify-ui (only time, BG and direction). 

When i load the watch, the simplify-ui works (but only some seconds, then the watch switch to "charging indicator") but by tipping on the screen, the simplify-ui comes back.

I tested all settings in the watch face configuration (Simplify UI: off, while charging, always on mode, always on and charging) when i use "while charging" or "always on and charging" the watch shows me the simplify-ui but only i loading the watch. I don't have the setting "always on" (i find it in the AAPS-Docs) in the watch face configuration.

It doesn't matter if i have the AOD on or off, it still doesn't work.

AAPS 3-2-0-4
AAPS Wear build at the same time as AAPS
Handy: Samsung Galaxy S23 ultra (Android 14 - OneUI 6.1)
Watch: Samsung Galaxy Watch 6 Classic (WearOS 4.0 - OneUI 5.0 Watch)

Philoul, 2024-11-13:
"Allways On" is not an AAPS or a Watchface parameter, but a watch parameter.

To find it slide down on watch to access to watch parameter, then click on icon below:
[image]

You can also access within Watch parameter > Screen > Always On Display
[image]

Tell me if it works with Always On Display enabled (be carefull, without charging, always on will reduce watch autonomy, so it's better to disable it during the day 😉)

MilosKozak, 2024-11-30:
closed for inactivity

closed 2024-11-30 by MilosKozak