nightscout/AndroidAPS #3444: How to send BG (from AAPS) through bluetooth ?

JohnDoeAkira, 2024-09-27:
Hello.
This is not a bug report but a help request.
I am flying a paraglider and I would like to try to send my BG value to my flying computer. On the flying computer side (an android tablet), I can set it up to receive BG value through bluetooth and display them on the screen to be able to monitor my sugar level in flight.
Is there a way, on AAPS, to send BG through bluetooth ?
Or would you know another way to do that ? (I use Dexcom G6 as sensors with BYODA).
Cheers,
Akira

andyrozman, 2024-09-27:
You could setup Xdrip to receive your G6 data (and then configure to broadcast data forward to AAPS), and then you could install Xdrip Follower on your "flying tablet", and configure it to send data to display data. 

If I am playing game and my android switches to gaming mode, I get my sugar displayed every 5 minutes on screen (but this is probably not good enough for you). You could write small android app that would be displayed over all apps (at screen edge and it could be receiving data from Xdrip Follower). Easiest way to see your BG is to have watch that displays that data, again Xdrip supports wide variety of them (they have even their own watch) so that would be an easy setup.

AAPS by itself doesn't transmit data anywhere, except to your watch if you have it configured. 

Without knowing more details of your current setup I can't give more detailed recommendations.

parapenT1sta, 2024-09-27:
Hi. I am also a paraglider pilot and I have xDrip sending glucose values to my MiBand 7 and has been working perfectly.
I had used G5 with xDrip. Now I am using Libre 2.

analogmonster, 2024-10-04:
BYODA has an option when building to broadcast to both AAPS and XDrip++. I mainly use BYODA and AAPS, the only reason I have XDrip++ is for one function: it can be set to say your bg and trend.

On my motorbike I cant as easily glance at a screen as when in my car. I have a bluetooth headset when on the bike and have XDrip++ speak the readings, slowly and twice. Only issue is if I am also trying to listen to my partner talk to me, or navigation notifications-which sound should take priority.

Another option is to have a high quality phone mount for your existing device running AAPS. I really like the RAM Mount quick grip for my motorbike. That way your flight computer is distraction free.

MilosKozak, 2024-11-01:
this not a task for "APS" system. Use xdrip

closed 2024-11-01 by MilosKozak

JohnDoeAkira, 2024-11-01:
Can I use Xdrip together with AAPS at the same time ?