nightscout/AndroidAPS #3447: suggestion: implement batching for uploads

cascer1, 2024-09-30:
AAPS always uploads data to nightscout as soon as it is available. Usually I think this is a good thing, but especially when my mobile data reception is poor this can eat through my battery quite quickly.

I'd love if there were some setting (in nsclient?) that's an intermediary because the options we have now. Uploading only on wifi isn't frequent enough for me, so maybe a slider for "when on mobile data, upload only once every x minutes". That way we can reduce the number of new connection attempts and thus I believe save some battery.

vanelsberg, 2024-10-01:
> I'd love if there were some setting 
I think it is already there. Did you try "Slow down uploads" (NSClient->Preferences->Advanced Settings)?

vanelsberg, 2024-10-01:
> eat through my battery quite quickly.

Btw: In case you are experiencing battery usage higher than expected, make sure to regular execute a "Database Cleanup".
You also may consider resetting the database. In some cases, database problems can cause high battery usage.

cascer1, 2024-10-03:
Oh look, slow down uploads does look like what I want. I can't find any more details in the documentation (other than the recommendation to leave it disabled). How does it work?

vanelsberg, 2024-10-07:
> Oh look, slow down uploads does look like what I want. I can't find any more details in the documentation (other than the recommendation to leave it disabled). How does it work?

When slow is enabled NDClient is delaying for 1 second between processing records (I think this lowers stressing the phones CPU and as such its power usage)

MilosKozak, 2024-11-01:
NSC v3 is the future and there will not be any difference in battery consumption

closed 2024-11-01 by MilosKozak