nightscout/AndroidAPS #3457: Frequent restarting

rpm5099, 2024-10-07:
Recently Android APS has been frequently restarting, and I assume crashing.  The time interval seems random but not more than 30 minutes.  This is occurring on my daughters phone, Oneplus 6T.  Nothing related to the restarts is being logged.  I have the same settings on my phone and have not had any issues for several years.  The version is 3.2.0.4 build b52015d35c-2024.05.30 compiled locally by myself.  I am also using the same version with the same configuration with no issues.  All permissions possible have been granted, it is not the Android kernel forcing the restarts.  The setup is using juggluco to connect to a Libre 3 sensor that pushes data to XDrip, which pushes to AAPS.  Seemingly related, the Nightscout server synchronization also seems to fail as well.  However after forcing the application to stop, clearing cache and restarting it then synchronizes.  This needs to be done multiple times per day.  

Thanks in advance for your help, I can provide any other information required.  

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Josch24, 2024-10-09:
Try "Clean Database" . I had similiar Behaviour some Time ago and it solved my Problem with crashing and Restarts.

rpm5099, 2024-10-09:
Thanks, I'll give that a shot although I think I did already.  I installed the debug version of the app but I still don't see anything related to these restarts in the logs at all.  It is really limiting to not be able to access the logs via SSH/USB, I can't ever check them remotely on my daughter's phone.

rpm5099, 2024-10-09:
Yeah, it didn't work, still lots of restart notifications.  I actually cleared the entire database and reinstalled the app from scratch a few days ago, although I restored the settings from a backup.  That does not seem to have helped either.  I've also verified the device is not running out of memory and I'm pretty certain the Android kernel is not killing the app as it has every permission it is possible to grant it.

rpm5099, 2024-10-13:
If anyone knows of any settings that can cause crashing/restarting, or any way to gather logs about what is happening I would appreciate any suggestions.  I've tried everything that I believe could reasonably be responsible.  Thanks.

Kaapeli64, 2024-10-27:
I think I'm having this same issue, symptoms at least match exactly. I set up logcat to follow AAPS PID and let it run until I get a restart. The last thing I saw in the logs was this:

10-27 19:45:16.068 24860  5289 W DynamiteLoaderV2Impl: Module config changed, forcing restart due to module measurement.dynamite
10-27 19:45:16.068 24860  5289 I Process : Sending signal. PID: 24860 SIG: 9

Anything before that appears to be normal operations. Any idea what that could be? Can we somehow avoid this forced restart? Do we know why it happens? Anything I can do to capture more verbose logs?

vanelsberg, 2024-10-27:
> 10-27 19:45:16.068 24860 5289 W DynamiteLoaderV2Impl: Module config changed, forcing restart due to module measurement.dynamite

I think this is what you see un the phone's logcat? It's not related of AAPS as far as I know.
That said, Google Play Services is known to kill apps for several reasons. So try disabling the "Google Services" system app it on this phone to see if this stops the restarting?

vanelsberg, 2024-10-29:
> How do I disable Google Services?

Settings->Apps (+select "show system apps)" -> "Google Play Services" -> "Disable"

vanelsberg, 2024-10-29:
> Reading here https://blogs.fsfe.org/larma/2017/signal-backdoors/ suggests this might be because AAPS is using some Google Services

Afaikt the issue is in the phone's OS. Not due to anything AAPS is doing. Some phones/brands just need the Google Play services to be disabled for AAPS working properly.

closed 2024-11-01 by MilosKozak

rpm5099, 2024-11-01:
@MilosKozak Did you mean to close this?  @vanelsberg and I are both still experiencing the restarts.  Is there any other troubleshooting steps we can try?

vanelsberg, 2024-11-01:
Nope. Not experiencing restarts.

MilosKozak, 2024-11-03:
There nothing we can do. Try another phone

Kaapeli64, 2024-11-03:
Yes, this most likely is a phone issue. What you could try is to collect adb logcat output and see what happened when the restart happened. There are a lot of things going on, so it might be helpful if you trace just prints coming from AAPS process ID. You can get the PID by running (on your PC, with your phone connected in debug mode):

adb shell "ps -A | grep info.nightscout.androidaps"

The once you find the PID, should be on the second column, eg 2531, you can just trace that one:

adb logcat --pid=2531

Leave it running until a restart. It might show something useful before it shuts down. It might also be something completely external to AAPS that shuts it down, in which case you would need to get the whole adb logcat instead and try to find the spot where the restart happened and see what other things happened there.

This might lead you closer to the actual issue what is going on.

rpm5099, 2024-11-03:
> s

@Kaapeli64 - Thanks, this is the kind of information I was looking for.  I do not have in depth technical experience with android, yet.  I will reply if I figure it out.