nightscout/AndroidAPS #3459: I added an insulin pump module, compiled and installed it on my mobile phone and reported an error: java. lang. VerifyError. Who can help see how to solve this problem?

kuang438951276, 2024-10-09:
FATAL EXCEPTION: main
                                                                                                    Process: info.nightscout.androidaps, PID: 26205
                                                                                                    java.lang.VerifyError: Superclass b.e of info.nightscout.androidaps.plugins.pump.eopatch.core.api.BuzzerStop is declared final (declaration of 'info.nightscout.androidaps.plugins.pump.eopatch.core.api.BuzzerStop' appears in /data/app/info.nightscout.androidaps-Vy5fpWG-LFJ6MXR2KKQspQ==/base.apk!classes2.dex)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.ble.PatchManagerImpl_Factory.newInstance(PatchManagerImpl_Factory.java:209)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.ble.PatchManagerImpl_Factory.get(PatchManagerImpl_Factory.java:152)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.ble.PatchManagerImpl_Factory.get(PatchManagerImpl_Factory.java:34)
                                                                                                    	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.ble.PatchManager_Factory.get(PatchManager_Factory.java:77)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.ble.PatchManager_Factory.get(PatchManager_Factory.java:19)
                                                                                                    	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.EopatchPumpPlugin_Factory.get(EopatchPumpPlugin_Factory.java:93)
                                                                                                    	at info.nightscout.androidaps.plugins.pump.eopatch.EopatchPumpPlugin_Factory.get(EopatchPumpPlugin_Factory.java:24)
                                                                                                    	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
                                                                                                    	at dagger.internal.MapFactory.get(MapFactory.java:58)
                                                                                                    	at dagger.internal.MapFactory.get(MapFactory.java:31)
                                                                                                    	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
                                                                                                    	at app.aaps.di.AppModule.providesPlugins(AppModule.kt:37)
                                                                                                    	at app.aaps.di.AppModule_ProvidesPluginsFactory.providesPlugins(AppModule_ProvidesPluginsFactory.java:77)
                                                                                                    	at app.aaps.di.DaggerAppComponent$AppComponentImpl.listOfPluginBase(DaggerAppComponent.java:31809)
                                                                                                    	at app.aaps.di.DaggerAppComponent$AppComponentImpl.injectMainApp(DaggerAppComponent.java:36081)
                                                                                                    	at app.aaps.di.DaggerAppComponent$AppComponentImpl.inject(DaggerAppComponent.java:36068)
                                                                                                    	at app.aaps.di.DaggerAppComponent$AppComponentImpl.inject(DaggerAppComponent.java:30082)
                                                                                                    	at dagger.android.DaggerApplication.injectIfNecessary(DaggerApplication.java:63)
                                                                                                    	at dagger.android.DaggerApplication.onCreate(DaggerApplication.java:38)
                                                                                                    	at app.aaps.MainApp.onCreate(MainApp.kt:95)
                                                                                                    	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1236)
                                                                                                    	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:7173)
                                                                                                    	at android.app.ActivityThread.access$1600(ActivityThread.java:276)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2175)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:107)
                                                                                                    	at android.os.Looper.loop(Looper.java:230)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8115)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:526)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1034)

MilosKozak, 2024-11-01:
reproduce the error with last dev or master version

MilosKozak, 2024-11-01:
what does mean "I added an insulin pump module"?

closed 2024-11-04 by MilosKozak