nightscout/AndroidAPS #3462: Tidepool 400 (bad request) error

sebastiaanvanderzee, 2024-10-12:
Hi, I found out that my data hasn't been uploaded to Tidepool from Aug 31 until now. I made a few screenshot. In one you can see a 'wrong' BG value of abour 60 mmol/l, could this be the reason Tidepool gives the 400 (bad request) error? I contacted Tidepool support, but they say they can't help me. I created another Tidepool account to see if I could upload there, but it didn't work. Is there a way to look at the data my phone is trying to upload, so I can manually remove some records?
--------------
- Build version AAPS 3.2.0.4.
[image]
[image]
[image]

vanelsberg, 2024-11-09:
Confirming: seeing this. 
AAPS 3.3.0-dev, from September 18th, 2024, Tidepool has no data. Same failure.

vanelsberg, 2024-11-12:
Response 400 shows invalid values for treatments/carbs.
It does not like AAPS's negative carbs. Will look into a fix.

closed 2024-11-30 by MilosKozak