nightscout/AndroidAPS #3466: Regression with BG type ??

robertrub, 2024-10-14:
I have problems with updating from a 2 weeks old dev to the last dev version. Seems like something went bad in a commit....

Build is ok, update is ok. AAPS craches systematically when starting. Clearing cache doesn't help

This is the logcat when the upgraded AAPS tries to start.

[image]

vanelsberg, 2024-10-14:
@MilosKozak 

I think the problem might be related to merge of **PR3411**:
It changes some contant values  (crashlogs complaining about not seeing the pre-PR values) So might only hit when running latest dev build with DexCom/xDrip?

Could database reset help out here?

Philoul, 2024-10-14:
@vanelsberg I think we can't request to all users of current master with xdrip + G6 to reset database before updating to 3.3...
Maybe we could keep key in sourceSensor enum class to avoid crash (probably easiest solution), or adapt converter, manage a database convertion on update...

plummer86, 2024-10-15:
I was thinking it could be handled by retuning an unknown sensor enum if the sourcesensor is not found.

Philoul, 2024-10-15:
Probably not manaed te same way wen you use enum to store an information in database, and wen you get an information from database...
But you are right, we should probably improve code to be more robust even when a database record doesn't match a key...

robertrub, 2024-10-15:
If db structure/names should be changed, then the code should read both entries but only write in the new format. That will clear the "old" format from the db in 6-9 months.

If not, a _db transformer code_ should change the db structure (always a risky business...) activated once, when it detects the old format.

robertrub, 2024-10-16:
Solved with the latest commit by @plummer86 
Closing this issue.

PR #3469 (2024-10-16): Handle unknown DB GlucoseValue SourceSensors
Fixes #3466

Updated toSourceSensor to return 'UNKNOWN' SourceSensor for GlucoseValues that have undefined sensor enum values. This change prevents toSourceSensor from raising an exception when the source sensor enum is not found.